Genesis-Embodied-AI/genesis-world #132: Gensis Version Conflict Error with Taichi on Mac Silicon M1

teccUI, 2024-12-20:
Hi Team, I'm getting the following errors on my Mac while installing Genesis. Attaching screenshots for reference:

[image]
[image]
[image]

Is there something that Im missing?

Cursky, 2024-12-20:
i find whl in https://pypi.tuna.tsinghua.edu.cn/simple/taichi to download whl
and change env to base , i successful install taichi 1.7.2

teccUI, 2024-12-20:
which version did you install? 
[image]
im getting this error lol

madrazzl3, 2024-12-20:
I had a similar issue, set up Python 3.12 and it's gone.

closed 2025-02-22 by YilingQiao