Genesis-Embodied-AI/genesis-world #134: mac运行出现Assertion failed:错误

dmhaomoon, 2024-12-20:
运行render_on_mamos.py脚本出现Assertion failed: (findOption(Name) == Values.size() && "Option already exists!"), function addLiteralOption, file CommandLine.h, line 846.错误，求大佬解答

FractalCatFX, 2024-12-20:
I am getting the same error:

>>> %Run -c $EDITOR_CONTENT
objc[60086]: Class GLFWHelper is implemented in both /Users/michael/anaconda3/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x163e00570) and /Users/michael/anaconda3/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x30d00d7c0). One of the two will be used. Which one is undefined.
objc[60086]: Class GLFWApplicationDelegate is implemented in both /Users/michael/anaconda3/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x163e005c0) and /Users/michael/anaconda3/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x30d00d810). One of the two will be used. Which one is undefined.
objc[60086]: Class GLFWWindowDelegate is implemented in both /Users/michael/anaconda3/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x163e005e8) and /Users/michael/anaconda3/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x30d00d838). One of the two will be used. Which one is undefined.
objc[60086]: Class GLFWContentView is implemented in both /Users/michael/anaconda3/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x163e00638) and /Users/michael/anaconda3/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x30d00d888). One of the two will be used. Which one is undefined.
objc[60086]: Class GLFWWindow is implemented in both /Users/michael/anaconda3/lib/python3.11/site-packages/taichi/_lib/core/taichi_python.cpython-311-darwin.so (0x163e006b0) and /Users/michael/anaconda3/lib/python3.11/site-packages/pygel3d/libPyGEL.dylib (0x30d00d900). One of the two will be used. Which one is undefined.
Assertion failed: (findOption(Name) == Values.size() && "Option already exists!"), function addLiteralOption, file CommandLine.h, line 846.

Process ended with exit code -6.

------

I tried uninstalling PyGEL3D to resolve the conflict, but Genesis seems to need both PyGEL3D and Taichi

FractalCatFX, 2024-12-21:
I had the same issue solved this by creating a custom environment in conda, and installing genesis fresh in that environment. Be sure to also install ffmpeg. Good luck!

closed 2024-12-27 by ziyanx02