Genesis-Embodied-AI/genesis-world #142: OpenGL.error.GLError: GLError(err = 1280；'NoneType' object is not callable

robot1lyj, 2024-12-20:
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/OpenGL/latebind.py", line 43, in __call__
    return self._finalCall( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/threading.py", line 1075, in _bootstrap_inner
    self.run()
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/threading.py", line 1012, in run
    self._target(*self._args, **self._kwargs)
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 640, in on_draw
    self._render()
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1086, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 916, in _update_context
    texture._add_to_context()
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/genesis/ext/pyrender/texture.py", line 219, in _add_to_context
    max_aniso = glGetFloat(texture_filter_anisotropic.GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/OpenGL/latebind.py", line 47, in __call__
    return self._finalCall( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/OpenGL/wrapper.py", line 678, in wrapperCall
    raise err
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/OpenGL/wrapper.py", line 671, in wrapperCall
    result = wrappedOperation( *cArguments )
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/lyj/anaconda3/envs/gen/lib/python3.12/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
	err = 1280,
	description = b'invalid enumerant',
	baseOperation = glGetFloatv,
	pyArgs = (
		GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
		<object object at 0x7f046d6640f0>,
	),
	cArgs = (
		GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
		array([0.], dtype=float32),
	),
	cArguments = (
		GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
		array([0.], dtype=float32),
	)
)

DavidJedeikin, 2024-12-20:
@robot1lyj what is your environment setup? Are attempting to runt he example from within a Docker container?

DavidJedeikin, 2024-12-20:
The issue seems to be caused by the latest version of PyOpenGL that is installed when running `pip install PyOpenGL`.  

Downgrading solved the issue for me. 
```
pip install PyOpenGL==3.1.1a1
```

Zechariah2001, 2024-12-21:
Exactly same error. I'm using MobaXterm to receive graphics from remote host, and have tried many versions of cuda, torch and PyOpenGL, still couldn't solve it. Environment setup:
```text
Cuda compilation tools, release 12.0, V12.0.76

NVIDIA-SMI 525.147.05   Driver Version: 525.147.05   CUDA Version: 12.0 

Python 3.12.0
nvidia-cuda-cupti-cu12   12.4.127
nvidia-cuda-nvrtc-cu12   12.4.127
nvidia-cuda-runtime-cu12 12.4.127
torch                    2.5.1
torchaudio               2.5.1
torchvision              0.20.1
PyOpenGL                 3.1.1a1
```

Xu-Xuanhui, 2024-12-21:
> Exactly same error. I'm using MobaXterm to receive graphics from remote host, and have tried many versions of cuda, torch and PyOpenGL, still couldn't solve it. Environment setup:
> 
> ```
> Cuda compilation tools, release 12.0, V12.0.76
> 
> NVIDIA-SMI 525.147.05   Driver Version: 525.147.05   CUDA Version: 12.0 
> 
> Python 3.12.0
> nvidia-cuda-cupti-cu12   12.4.127
> nvidia-cuda-nvrtc-cu12   12.4.127
> nvidia-cuda-runtime-cu12 12.4.127
> torch                    2.5.1
> torchaudio               2.5.1
> torchvision              0.20.1
> PyOpenGL                 3.1.1a1
> ```
PyOpenGL does not support Python versions 3.10 and above well. It is recommended to install Python version 3.10.

xiaobai12413, 2024-12-24:
安装 Python 版本 3.10还是不行

xiaobai12413, 2024-12-24:
是因为我的虚拟机没有gpu吗

duburcqa, 2025-02-28:
Fixed by https://github.com/Genesis-Embodied-AI/Genesis/pull/777

closed 2025-02-28 by duburcqa