Genesis-Embodied-AI/genesis-world #143: AttributeError: function 'glGetUniformLocation' not found

mayurshri, 2024-12-20:
Getting this error after running test.py.

import genesis as gs
gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()

This is what it says

 File "E:\Genesis-AI\test.py", line 10, in <module>
    scene.build()
  File "C:\Users\may_s\lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "C:\Users\may_s\lib\site-packages\genesis\engine\scene.py", line 556, in build
    self._visualizer.build()
  File "C:\Users\may_s\lib\site-packages\genesis\vis\visualizer.py", line 101, in build
    self._context.build(self._scene)
  File "C:\Users\may_s\lib\site-packages\genesis\vis\rasterizer_context.py", line 70, in build
    self.jit = JITRenderer(self._scene, [], [])
  File "C:\Users\may_s\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in __init__
    self.gen_func_ptr()
  File "C:\Users\may_s\lib\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
    self.gl = GLWrapper()
  File "C:\Users\may_s\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 24, in __init__
    load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
  File "C:\Users\may_s\lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 52, in load_func
    func_ptr = GL.platform.ctypesloader.buildFunction(
  File "C:\Users\may_s\lib\site-packages\OpenGL\platform\ctypesloader.py", line 96, in buildFunction
    return functionType( (name, dll), )
AttributeError: function 'glGetUniformLocation' not found

lyraestar, 2024-12-20:
I also get this error

manuelgardina, 2024-12-20:
yep, same

LucaSalesi98, 2024-12-20:
the same here with windows 11

zhouxian, 2024-12-22:
What OS are you on?

tinypumpkin, 2024-12-22:
Traceback (most recent call last):
  File "f:\genesis\temp.py", line 12, in <module>
    scene.build()
  File "D:\Anaconda\Lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Anaconda\Lib\site-packages\genesis\engine\scene.py", line 556, in build
    self._visualizer.build()
  File "D:\Anaconda\Lib\site-packages\genesis\vis\visualizer.py", line 101, in build
    self._context.build(self._scene)
  File "D:\Anaconda\Lib\site-packages\genesis\vis\rasterizer_context.py", line 70, in build
    self.jit = JITRenderer(self._scene, [], [])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Anaconda\Lib\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in __init__
    self.gen_func_ptr()
  File "D:\Anaconda\Lib\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
    self.gl = GLWrapper()
              ^^^^^^^^^^^
  File "D:\Anaconda\Lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 24, in __init__
    load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
  File "D:\Anaconda\Lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 52, in load_func
    func_ptr = GL.platform.ctypesloader.buildFunction(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "D:\Anaconda\Lib\site-packages\OpenGL\platform\ctypesloader.py", line 96, in buildFunction
    return functionType( (name, dll), )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: function 'glGetUniformLocation' not found ，I got this

tinypumpkin, 2024-12-22:
win11 , linux is ok ？

simlabsoft, 2024-12-25:
Same on Windows 11

ziyanx02, 2024-12-27:
Close as a duplicate. Please refer to #12 for more information.

closed 2024-12-27 by ziyanx02

walterf25, 2025-01-02:
I also get the same error when trying to run every example, only if I set the following to true
`show_viewer=True`
I'm running on windows 10, python 3.10, is there a work around for this?

hadisaab2, 2025-01-06:
same error

ehsanecc, 2025-01-15:
Same, [here](https://genesis-world.readthedocs.io/en/latest/user_guide/overview/installation.html#prerequisites) they stated that **Interactive Viewer** is not available on windows so `show_viewer` should be `False` for windows.