Genesis-Embodied-AI/genesis-world #145: [Bug]  run  hello_genesis.py  meet visualizer crash

AugustChen-yjh, 2024-12-20:
Hi， I met this crash when I run hello_genesis.py
Here is my computer configurations:
```
Hardware:
CPU : AMD Ryzen 9 7950X 16-Core Processor
GPU : NVIDIA RTX A6000 
MEM : 64GB DDR5

Software:
OS: Ubuntu 22.04
python-venv : annaconda + python3.10 + torch
```
and I run command like this:
```
cd ~/workspace/Genesis/examples/tutorials
python hello_genesis.py
```
then it output like this
```bash
[Genesis] [15:46:23] [INFO] Building visualizer...
libGL error: MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open radeonsi: /usr/lib/dri/radeonsi_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: radeonsi
libGL error: MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
libGL error: failed to load driver: swrast
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/home/cz/anaconda3/envs/genesis/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/home/cz/anaconda3/envs/genesis/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/cz/workspace/Genesis/genesis/ext/pyrender/viewer.py", line 1126, in _init_and_start_app
    super(Viewer, self).__init__(
  File "/home/cz/anaconda3/envs/genesis/lib/python3.10/site-packages/pyglet/window/xlib/__init__.py", line 168, in __init__
    super().__init__(*args, **kwargs)
  File "/home/cz/anaconda3/envs/genesis/lib/python3.10/site-packages/pyglet/window/__init__.py", line 527, in __init__
    context = config.create_context(gl.current_context)
  File "/home/cz/anaconda3/envs/genesis/lib/python3.10/site-packages/pyglet/gl/xlib.py", line 114, in create_context
    return XlibContext(self, share)
  File "/home/cz/anaconda3/envs/genesis/lib/python3.10/site-packages/pyglet/gl/xlib.py", line 149, in __init__
    raise gl.ContextException(msg)
pyglet.gl.ContextException: Could not create GL context
^CTraceback (most recent call last):
  File "/home/cz/workspace/Genesis/examples/tutorials/hello_genesis.py", line 18, in <module>
    scene.build()
  File "/home/cz/workspace/Genesis/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/home/cz/workspace/Genesis/genesis/engine/scene.py", line 565, in build
    self._visualizer.build()
```
I think this may caused by AMD driver and Nvidia driver mistmatch, will you have any batter idea?

zyz-code, 2024-12-20:
https://github.com/Genesis-Embodied-AI/Genesis/issues/119

RRGGZZ, 2024-12-20:
I also encountered the same problem, I was the first few times the program did not report errors and ran normally, but the window came out with a black screen and no display, and then I ran several times before this problem began to appear

AugustChen-yjh, 2024-12-20:
> #119

Thanks , I have try it out , but it dose not work for me. They are requiring different driver files link

RRGGZZ, 2024-12-20:
My wsl2 ubuntu24.04.01 system, my cpu is intel and gpu is Nvidia also have this problem, so I think we should wait for others to see the situation

It may cuased by AMD cpus within GPUs ,should we install AMD RComm drivers from website?

AugustChen-yjh, 2024-12-20:
Hi guys! I've worked it out !
run commnad like this 
```
mkdir -p /usr/lib/dri/
ln -s /usr/lib/x86_64-linux-gnu/dri/swrast_dri.so /usr/lib/dri/swrast_dri.so

conda install -c conda-forge gcc
```
and it works!
[image]

RRGGZZ, 2024-12-20:
thanks！ I‘ll try it again！

From：
 TYUT - 任冠竹（renguanzhu）
Class And Grade：H2104
Student Number：2021008008

_1734683560292.png (view on web)<https://github.com/user-attachments/assets/ca9158f1-c162-4e36-8c30-0eb567b4541e>