Genesis-Embodied-AI/genesis-world #146: 很多能跑起来的案例也很卡顿啊，有些是没有界面显示但是终端在跑，有没有像文生图那种保存视频的看看的

zyz-code, 2024-12-20:


Wu-Fisher, 2024-12-20:
“有些是没有界面显示但是终端在跑” Add -v to the startup parameters and It should work. "文生图那种保存视频" to #30

zyz-code, 2024-12-20:
> “有些是没有界面显示但是终端在跑” Add -v to the startup parameters and It should work. "文生图那种保存视频" to #30

跑之前看了一下代码的，需要的我加了的，不需要的没加

zyz-code, 2024-12-20:
> “有些是没有界面显示但是终端在跑” Add -v to the startup parameters and It should work. "文生图那种保存视频" to #30

thanks

ziyanx02, 2024-12-21:
Could you please list some examples that should have displayed a viewer but failed in your terminal?

zyz-code, 2024-12-22:
> Could you please list some examples that should have displayed a viewer but failed in your terminal?

就是终端这边在提示多少fps，最后一直到结束那个显示界面都一直是黑的，正常像那个龙、碰撞案例都是有个简单动画嘛，对这个我也是外行，暂时不弄这个了，真有啥好的展示可以踢我一下

ziyanx02, 2024-12-22:
We greatly appreciate your interest in Genesis! Access to the generative framework will be rolled out gradually in the near future. We will continue to provide updates in #6, so stay tuned!

closed 2024-12-22 by ziyanx02