Genesis-Embodied-AI/genesis-world #147: As suggested in /issues/10

lambooking, 2024-12-20:
As suggested in /issues/10
Please give it a try:
```
PYOPENGL_PLATFORM=osmesa python try.py 
```

_Originally posted by @zhenjia-xu in https://github.com/Genesis-Embodied-AI/Genesis/issues/37#issuecomment-2553118072_

lambooking, 2024-12-20:
Traceback (most recent call last):
  File "/home/lem5/code/genesis/examples/tutorials/hello_genesis.py", line 1, in <module>
    import genesis as gs
  File "/home/lem5/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/__init__.py", line 274, in <module>
    from .engine.scene import Scene
  File "/home/lem5/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/engine/scene.py", line 5, in <module>
    from genesis.engine.entities import Emitter
  File "/home/lem5/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/engine/entities/__init__.py", line 1, in <module>
    from .avatar_entity import AvatarEntity
  File "/home/lem5/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/engine/entities/avatar_entity/__init__.py", line 1, in <module>
    from .avatar_entity import AvatarEntity
  File "/home/lem5/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/engine/entities/avatar_entity/avatar_entity.py", line 3, in <module>
    from ..rigid_entity import RigidEntity
  File "/home/lem5/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/engine/entities/rigid_entity/__init__.py", line 1, in <module>
    from .rigid_entity import RigidEntity
  File "/home/lem5/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/engine/entities/rigid_entity/rigid_entity.py", line 9, in <module>
    from genesis.utils import mjcf as mju
  File "/home/lem5/anaconda3/envs/genesis/lib/python3.10/site-packages/genesis/utils/mjcf.py", line 3, in <module>
    import mujoco
  File "/home/lem5/anaconda3/envs/genesis/lib/python3.10/site-packages/mujoco/__init__.py", line 57, in <module>
    from mujoco.gl_context import *
  File "/home/lem5/anaconda3/envs/genesis/lib/python3.10/site-packages/mujoco/gl_context.py", line 41, in <module>
    from mujoco.egl import GLContext as _GLContext
  File "/home/lem5/anaconda3/envs/genesis/lib/python3.10/site-packages/mujoco/egl/__init__.py", line 26, in <module>
    raise ImportError(
ImportError: Cannot use EGL rendering platform. The PYOPENGL_PLATFORM environment variable is set to {!r} (should be either unset or 'egl').

duburcqa, 2025-03-22:
Rendering issues have been fixed on main branch.

closed 2025-03-22 by duburcqa