Genesis-Embodied-AI/genesis-world #148: how can I record the video of examples/smoke.py

Carlofkl, 2024-12-20:


arielhgl, 2024-12-22:
the sim generates a lists of images in a created directory, after the images are generated you can run the following:
`ffmpeg -framerate 30 -i examples/video/%04d.png -c:v libx264 -pix_fmt yuv420p output.mp4`

this should create a video of the generated images.

arielhgl, 2024-12-22:
https://github.com/user-attachments/assets/99a46c6c-c5cd-4572-8e83-fe0440bfe769

this was the result for me.

Carlofkl, 2024-12-23:
> the sim generates a list of images in a created directory, after the images are generated you can run the following: `ffmpeg -framerate 30 -i examples/video/%04d.png -c:v libx264 -pix_fmt yuv420p output.mp4`
> 
> this should create a video of the generated images.
> 
> Make sure you point the source image path to the right location.

actually how can I record a video with different views from cameras, (the 3d version video)

Kashu7100, 2024-12-26:
You can also use `gs.tools.animate` for easier saving. 

```python
frames = []
for i in range(100):
    scene.step()
    res = cam.render()
    frames.append(res[0])
gs.tools.animate(frames, "demo.mp4")
```

closed 2025-02-22 by YilingQiao