Genesis-Embodied-AI/genesis-world #150: AttributeError: partially initialized module 'genesis' has no attribute 'generate' (most likely due to a circular import)

clairelee5740, 2024-12-20:
I found this project amazing and hope to try out as in the demo video

here is my code:
```
import genesis as gs

gs.generate("A water droplet drops onto a beer bottle ans then slowly slides down along the bottle surface.")
```

However, it says:
```
Traceback (most recent call last):
  File "C:\genesis.py", line 1, in <module>
    import genesis as gs
  File "C:\genesis.py", line 3, in <module>
    gs.generate("A water droplet drops onto a beer bottle ans then slowly slides down along the bottle surface.")
AttributeError: partially initialized module 'genesis' has no attribute 'generate' (most likely due to a circular import)
```

I have installed genesis-world

Could anyone solve the problem? Thanks!

clairelee5740, 2024-12-20:
OK I found in #30 that the generate function is not released yet, as in the readme it says 

> Some of the modules integrated existing papers and some are still under submission. Access to our generative feature will be gradually rolled out in the near future

So currently, what can Genesis do now???

zhouxian, 2024-12-20:
Hi, thank you for your interest! Genesis's upper level generative system is a modular framework consisting of multiple modules, and is not ready for public access yet. As of this stage, we are only releasing the low level physics engine.

jzc1994, 2024-12-20:
same question！

ziyanx02, 2024-12-22:
We will continue to provide updates about the generative framework in #6, so stay tuned!

closed 2024-12-22 by ziyanx02