Genesis-Embodied-AI/genesis-world #153: when i used system dependencies (2.A) to build LuisaRender and its python binding it have some problem

HANQAQ, 2024-12-20:
when i use:`cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.9 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON -D LUISA_COMPUTE_ENABLE_GUI=OFF `
it show:--```
 Found pybind11: /home/han/mambaforge/envs/genesis/include (found version "2.13.6")
-- Configuring done (0.7s)
-- Generating done (0.3s)
-- Build files have been written to: /home/han/Genesis/genesis/ext/LuisaRender/build
```
then 
i use:`cmake --build build -j $(nproc)`
it show:```
warning: `luisa_compute_backend_impl` (lib) generated 10 warnings
error: could not compile `luisa_compute_backend_impl` (lib) due to 4 previous errors; 10 warnings emitted
make[2]: *** [src/compute/src/rust/CMakeFiles/luisa_compute_rust_build_embree.dir/build.make:147: src/compute/src/rust/rust/release/libluisa_compute_ir_static.a] Error 101
make[2]: *** Deleting file 'src/compute/src/rust/rust/release/libluisa_compute_ir_static.a'
make[1]: *** [CMakeFiles/Makefile2:2186: src/compute/src/rust/CMakeFiles/luisa_compute_rust_build_embree.dir/all] Error 2
[ 46%] Building CXX object src/compute/src/osl/CMakeFiles/luisa-compute-osl.dir/oso_parser.cpp.o
/usr/bin/ld: CMakeFiles/luisa-cuda-nvrtc-standalone-compiler.dir/cuda_nvrtc_compiler.cpp.o: in function `main':
cuda_nvrtc_compiler.cpp:(.text.startup+0x173): undefined reference to `nvrtcGetOptiXIRSize'
/usr/bin/ld: cuda_nvrtc_compiler.cpp:(.text.startup+0x197): undefined reference to `nvrtcGetOptiXIR'
collect2: error: ld returned 1 exit status
make[2]: *** [src/compute/src/backends/cuda/CMakeFiles/luisa-cuda-nvrtc-standalone-compiler.dir/build.make:104: bin/luisa_nvrtc] Error 1
make[1]: *** [CMakeFiles/Makefile2:2576: src/compute/src/backends/cuda/CMakeFiles/luisa-cuda-nvrtc-standalone-compiler.dir/all] Error 2
[ 46%] Linking CXX shared library ../../../../bin/liblc-ir-v2.so
[ 46%] Built target luisa-compute-ir-v2
[ 46%] Linking CXX shared library ../../../../bin/liblc-tensor.so
[ 46%] Built target luisa-compute-tensor
[ 46%] Linking CXX shared library ../../../../bin/liblc-osl.so
[ 46%] Built target luisa-compute-osl
make: *** [Makefile:136: all] Error 2
```
how i solve this problem

zswang666, 2024-12-20:
It looks like you don't have or have the wrong cuda runtime compilation library
```
usr/bin/ld: CMakeFiles/luisa-cuda-nvrtc-standalone-compiler.dir/cuda_nvrtc_compiler.cpp.o: in function `main':
cuda_nvrtc_compiler.cpp:(.text.startup+0x173): undefined reference to `nvrtcGetOptiXIRSize'
/usr/bin/ld: cuda_nvrtc_compiler.cpp:(.text.startup+0x197): undefined reference to `nvrtcGetOptiXIR'
```
Please check the [FAQ](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html#faq) and see if that fixes the issue.

HANQAQ, 2024-12-22:
> It looks like you don't have or have the wrong cuda runtime compilation library
>
> ```
> usr/bin/ld: CMakeFiles/luisa-cuda-nvrtc-standalone-compiler.dir/cuda_nvrtc_compiler.cpp.o: in function `main': [...]

it maybe i have not OptiX sdk?

ACMLCZH, 2024-12-26:
To support functions like `nvrtcGetOptiXIRSize`, you should update your CUDA Toolkit to version 12.0 or higher. The link in the document will be updated soon to NVIDIA's CUDA 12.1.0 download archive.

Additionally, Issues #275 and #194 might be helpful to ensure your CUDA Toolkit is correctly installed.

HANQAQ, 2024-12-26:
> To support functions like `nvrtcGetOptiXIRSize`, you should update your CUDA Toolkit to version 12.0 or higher. The link in the document will be updated soon to NVIDIA's CUDA 12.1.0 download [...]

OK, i got it！ Thanks a lot !  :)

closed 2025-02-23 by YilingQiao