Genesis-Embodied-AI/genesis-world #154: Config bugs when running `python examples/locomotion/go2_train.py`

DuinoDu, 2024-12-20:
Hi, thanks for great job!
I try to execute `python examples/locomotion/go2_train.py` but get much bugs about on_policy_runner config bug. Is this feature under developing?

ziyanx02, 2024-12-21:
Could you let us know the version of `rsl_rl` you are using? 

There is a significant difference between 2.0.1 (automatically installed via `pip install rsl_rl`) and 1.0.2

SuanCai-Fish, 2024-12-22:
> Could you let us know the version of `rsl_rl` you are using?您能否告诉我们您正在使用的`rsl_rl`版本？
> 
> There is a significant difference between 2.0.1 (automatically installed via `pip install rsl_rl`) and 1.0.22.0.1（通过`pip install rsl_rl`自动安装）和 1.0.2 之间存在显着差异

> Hi, thanks for great job!您好，感谢您的出色工作！ I try to execute `python examples/locomotion/go2_train.py` but get much bugs about on_policy_runner config. Is this feature under developing?我尝试执行 `python examples/locomotion/go2_train.py` 但关于 on_policy_runner 配置有很多错误。该功能正在开发中吗？

I'm a beginner for learning RL. I follow the User Guide in "Training Locomotion Policies with RL". Is the only thing I need to do is to "Install rsl_rl"? or there is something i forget to do? I can't seem to find a file named "go2_train.py. Could you let me know which step i forget to do?

ziyanx02, 2024-12-22:
> I'm a beginner for learning RL. I follow the User Guide in "Training Locomotion Policies with RL". Is the only thing I need to do is to "Install rsl_rl"? or there is something i forget to do? I [...]

Yes, the only thing you need to do is to install rsl_rl (v1.0.2) from here: https://github.com/leggedrobotics/rsl_rl/releases/tag/v1.0.2
Please ensure you are in `examples/locomotion` directory before trying to run `go2_train.py`.

If you encounter any other issues, feel free to open a new one to let us know.

ziyanx02, 2024-12-22:
Here is another rl example for Genesis installed by `pip install genesis-world`:
https://github.com/ziyanx02/Genesis-backflip

closed 2024-12-22 by ziyanx02

Nate711, 2025-04-10:
I have the same issue even using v1.0.2