Genesis-Embodied-AI/genesis-world #155: How does VLM interact with physics simulation engines?

KeliangLiu2023, 2024-12-20:
Hello, I would like to ask what role VLM played in this project? How does VLM interact with physics simulation engines? I don't think I saw that part of the supporting paper.

ziyanx02, 2024-12-28:
Please follow #6 for updates on the generative framework. VLM will be included in the following works.
I will close this issue as it is not related to the simulator.

closed 2024-12-28 by ziyanx02

BurhanUlTayyab, 2025-01-01:
Understand that you guys will release it later. But we are more interested in understanding how VLM interacts with the simulator.

zhouxian, 2025-01-01:
> Understand that you guys will release it later. But we are more interested in understanding how VLM interacts with the simulator.

Yeah sure -- the idea is actually quite simple: we made the APIs simple enough so that it's understandable by LLM, then we just ask LLM to write code. The capability for writing simulation code is way below the limit of today's LLM's coding ability.

There are also of course other modules involved, but the basic idea is instead of learning everything end-to-end with a gigantic neural network, we generate stuff by using physics engine as a tool and simply write code