Genesis-Embodied-AI/genesis-world #156: [ERROR] cuda device not available

krenax, 2024-12-20:
I get this error when running the complete example of the section Control your Robot.

analogjedi, 2024-12-20:
@krenax , this is most likely not a Genesis project issue. Would you please provide your environment details (OS, Conda/Venv config, etc. )?

krenax, 2024-12-20:
My machine is running Linux Mint 21.3 x86_64 with an AMD Radeon RX 7900 XTX. Which config would be helpful?

analogjedi, 2024-12-20:
@krenax , Cuda is an Nvidia GPU framework. Unfortunately, your AMD GPU is not compatible with Cuda.

krenax, 2024-12-20:
Perhaps I do misunderstand the description, but: The project description states that this project is cross-platform and runs across different compute backend (AMD GPU included).

computer-rookie-master, 2024-12-21:
hi, Nvidia Gpus contain multiple cuda units, if you have a nvidia gpu, gs.cuda() works. AMDs should find another way

duburcqa, 2025-03-22:
@krenax You should try with Vulkan backend, which should be supported. This is now the default behaviour on main branch if Cuda is not available.

closed 2025-03-22 by duburcqa