Genesis-Embodied-AI/genesis-world #157: Musculoskeletal Bodies Simulation

utkuoguzman, 2024-12-20:
Hi,
When you state "Genesis supports simulation (and the coupling) of rigid and articulated bodies", is this also valid for musculoskeletal bodies, meaning the muscle simulation of MuJoCo and OpenSim and Hyfydy? It is a big part of the MuJoCo community. If this feature is not included, can you also put the musculoskeletal bodies simulation into your backlogs?
Thanks.

JiangJubo, 2024-12-20:
Similarly, I am also concerned about whether the environment supports MuJoCo tendon geom, which makes up most musculoskeletal models in MuJoCo.
Thanks.

Albertchamberlain, 2024-12-23:
yes, need human body simulation

Vittorio-Caggiano, 2024-12-23:
would love to learn more about the MuJoCo tendon support, ideally to support [MyoSuite task suites](https://github.com/MyoHub/myosuite) :) 
In the meanwhile I had a test with human physiological skeletal simulation from [myoskeleton](https://github.com/myolab/myo_model) and on an M3 Max with 36 GB run at 50 fps (which seem a bit slow). Did anybody have similar observations for models of similar complexity?

zhouxian, 2025-01-02:
> would love to learn more about the MuJoCo tendon support, ideally to support [MyoSuite task suites](https://github.com/MyoHub/myosuite) :) In the meanwhile I had a test with human physiological [...]

If you do not need parallel simulation, using ‘backend=gs.cpu’ would greatly improve speed

also rendering could be the bottleneck. There’s still part of the rendering pipeline (specifically scene state update) that is still in raw python and we haven’t jitted. You could turn off rendering and check the speed again. If it’s still slow, do let us know

hudsonburke, 2025-01-02:
> would love to learn more about the MuJoCo tendon support, ideally to support [MyoSuite task suites](https://github.com/MyoHub/myosuite) :) In the meanwhile I had a test with human physiological [...]

Did you have to edit the myoskeleton files to get it to show up? I tried this one and several other's from myosim, but it would either error out or just show a blank plane like in one of the issues

Hippozhibos, 2025-01-04:
Hi developers,

I’ve been following the discussions about muscle and tendon modeling (e.g., #374 and #287), and I noticed many users are interested in these features, but they’re currently not supported in Genesis.

Given that this seems like a valuable extension for the framework, I was wondering if it would be possible for the community to contribute towards implementing these features. Specifically:
1. Do you think it's feasible to add muscle and tendon simulation support to Genesis?
2. Are there any suggestions on how we could start building these features, possibly within the existing codebase?

I’d love to hear your thoughts, and if anyone from the community is interested in joining the effort, that would be great!

Thanks for your time and guidance.

Vittorio-Caggiano, 2025-01-04:
> Did you have to edit the myoskeleton files to get it to show up? I tried this one and several other's from myosim, but it would either error out or just show a blank plane like in one of the issues

@hudsonburke  yes, I had to remove `slide` joints to make it work

Vittorio-Caggiano, 2025-01-04:
> If you do not need parallel simulation, using ‘backend=gs.cpu’ would greatly improve speed
>
> also rendering could be the bottleneck. There’s still part of the rendering pipeline (specifically scene [...]

thanks @zhouxian when I remove rendering the performance increase to ~620 FPS (M3 mac Max). Will try to do some test on parallel as well.