Genesis-Embodied-AI/genesis-world #158: How to install LuisaRender ?

Xu-Xuanhui, 2024-12-20:
I’ve been trying to install LuisaRender for the past two days, but it seems like I’m still far from success. I hope to receive guidance from professionals.

My system is Nvidia 4070 Laptop, Ubuntu 22.04, Nvidia driver 535.183.01, CUDA 11.8, Python 3.10.

Here are my attempts:

First, I compiled LuisaRender from https://github.com/LuisaGroup/LuisaRender. However, after compiling, there was no Python interface and I couldn’t import LuisaRenderPy.

After this, I realized it was an issue with the version of LuisaRender. A professional told me I needed the correct fork of LuisaRender. So, I found the Alif-01 fork. This time, the compilation was successful, and I could import LuisaRenderPy. However, a new error occurred: "I cannot find the 'create_scene' API in LuisaRenderPy's pybind API code." I tried print(dir(LuisaRenderPy)), and the output was:

['Camera', 'Integrator', 'LogLevel', 'Shape', 'Spectrum', 'Surface', 'Texture', 'Transform', 'doc', 'file', 'loader', 'name', 'package', 'spec', 'add_emission', 'add_environment', 'add_surface', 'destroy', 'init', 'render_frame', 'update_camera', 'update_shape']
It seems that LuisaRenderPy doesn’t have 'create_scene'?

Next, I looked at the Genesis User Guide. Firstly, I didn’t see LuisaRender in the genesis/ext folder. Secondly, the libraries mentioned in the guide are for LuisaCompute. I tried creating a LuisaRender folder under genesis/ext, then ran git clone https://github.com/LuisaGroup/LuisaCompute.git. After installing all dependencies, I began compiling, but encountered a new issue:

[ 56%] Built target luisa-compute-backend-remote
/usr/bin/ld: /tmp/ccYrDqwJ.ltrans0.ltrans.o: in function `main':
<artificial>:(.text.startup+0x17b): undefined reference to `nvrtcGetOptiXIRSize'
/usr/bin/ld: <artificial>:(.text.startup+0x19f): undefined reference to `nvrtcGetOptiXIR'
collect2: error: ld returned 1 exit status
gmake[2]: *** [src/backends/cuda/CMakeFiles/luisa-cuda-nvrtc-standalone-compiler.dir/build.make:104: bin/luisa_nvrtc] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1789: src/backends/cuda/CMakeFiles/luisa-cuda-nvrtc-standalone-compiler.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
This seems to indicate that NVIDIA-OptiX is not installed.

As a beginner in ray tracing rendering, I’m feeling a bit lost at this point.

The emergence of Genesis has excited me greatly. A good physics engine is an indispensable tool for robotics researchers. Ray tracing brings displays closer to reality, making it an excellent tool for showcasing work results and for vision-based robotic control. I hope the official team will release a detailed installation guide for LuisaRender, as this would greatly benefit the community’s growth.

Xu-Xuanhui, 2024-12-20:
I have install the LuisaRender
Ubuntu 22.04 Nvidia driver 535.183.01,cuda 12.2! I update the cuda! python 3.10
git clone https://github.com/Genesis-Embodied-AI/Genesis.git
cd Genesis
git submodule update --init --recursive

You should follow the user guider to install the Dependencies.

cd genesis/ext/LuisaRender
cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.10 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON -D LUISA_COMPUTE_ENABLE_GUI=OFF 
cmake --build build -j $(nproc)

Finall I copy the etx folder to my conda envs.
It works!

zswang666, 2024-12-20:
Hey thanks for sharing the guide and sorry for the delay on this part of the doc!

We have the full guide [here](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html#photo-realistic-ray-tracing-rendering).

closed 2024-12-20 by zswang666

Varin6, 2024-12-29:
@Xu-Xuanhui 

> Finall I copy the etx folder to my conda envs.

What do you mean by this? Do you just add the path to PATH? Or do you physically copy the folder with the built LuisaRender? 

I managed to build mine but when I try running demo.py, I'm still getting :

`[Genesis] [02:51:28] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: 0, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Traceback (most recent call last):
  File "/home/varin/.local/lib/python3.12/site-packages/genesis/vis/raytracer.py", line 17, in <module>
    import LuisaRenderPy
ModuleNotFoundError: No module named 'LuisaRenderPy'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/varin/Dev/Genesis/examples/rendering/demo.py", line 166, in <module>
    main()
  File "/home/varin/Dev/Genesis/examples/rendering/demo.py", line 11, in main
    scene = gs.Scene(
            ^^^^^^^^^
  File "/home/varin/.local/lib/python3.12/site-packages/genesis/utils/misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "/home/varin/.local/lib/python3.12/site-packages/genesis/engine/scene.py", line 148, in __init__
    self._visualizer = Visualizer(
                       ^^^^^^^^^^^
  File "/home/varin/.local/lib/python3.12/site-packages/genesis/vis/visualizer.py", line 60, in __init__
    from .raytracer import Raytracer
  File "/home/varin/.local/lib/python3.12/site-packages/genesis/vis/raytracer.py", line 19, in <module>
    gs.raise_exception(f"Failed to import LuisaRenderer. {e.__class__.__name__}: {e}")

[Genesis] [02:51:31] [ERROR] Failed to import LuisaRenderer. ModuleNotFoundError: No module named 'LuisaRenderPy'
`

kaspar-gestalt, 2025-01-02:
@Varin6 I think you might have to add the path to the the .so file (e.g. LuisaRenderPy.cpython-39-x86_64-linux-gnu.so) to the PYTHONPATH: `export PYTHONPATH=...:$PYTHONPATH`