Genesis-Embodied-AI/genesis-world #159: Trimesh Error Report

Yzhang367, 2024-12-20:
```terminal
Traceback (most recent call last):
  File "/home/yzhang/桌面/VSCode_Py/genesis/test_1.py", line 3, in <module>
    scene = gs.Scene(
            ^^^^^^^^^
  File "/home/yzhang/.local/lib/python3.12/site-packages/genesis/utils/misc.py", line 27, in new_init
    original_init(self, *args, **kwargs)
  File "/home/yzhang/.local/lib/python3.12/site-packages/genesis/engine/scene.py", line 148, in __init__
    self._visualizer = Visualizer(
                       ^^^^^^^^^^^
  File "/home/yzhang/.local/lib/python3.12/site-packages/genesis/vis/visualizer.py", line 26, in __init__
    self._context = RasterizerContext(vis_options)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yzhang/.local/lib/python3.12/site-packages/genesis/vis/rasterizer_context.py", line 37, in __init__
    self.init_meshes()
  File "/home/yzhang/.local/lib/python3.12/site-packages/genesis/vis/rasterizer_context.py", line 52, in init_meshes
    self.link_frame_mesh = trimesh.creation.axis(origin_size=0.03, axis_radius=0.025, axis_length=1.0)
                           ^^^^^^^
NameError: name 'trimesh' is not defined
```
I came across this message many times and tried `pip install trimesh` then put `import trimesh` in `rasterizer_context.py` but it didn't work. I was quite upset.

analogjedi, 2024-12-20:
I had the same issue on my Mac M3 and it was related to this issue: #26

DuinoDu, 2024-12-20:
Genesis doesn't use trimesh from pip install, but from `genesis/ext/trimesh`. See [here](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/genesis/vis/rasterizer_context.py#L8). You can check by 
```
from genesis.ext import pyrender, trimesh
from genesis.ext.pyrender.jit_render import JITRenderer
```
to figure out why it is not imported.

Yzhang367, 2024-12-21:
> Genesis doesn't use trimesh from pip install, but from `genesis/ext/trimesh`. See [here](https://github.com/Genesis-Embodied-AI/Genesis/blob/main/genesis/vis/rasterizer_context.py#L8). You can check [...]

> I had the same issue on my Mac M3 and it was related to this issue: #26

Thanks so much for your replies. The problem was solved when I perform the following steps referring to #26.
1. Pip install `trimesh` and `pyrender`
2. Turning the origin `try-except` to
```python
try:
     from genesis.ext import pyrender, trimesh
     from genesis.ext.pyrender.jit_render import JITRenderer
except:
    import trimesh
    import pyrender
``` 
Then the error message turned into OpenGL error. To solve it, I use `pip install -U PyOpenGL` to upgrade it to the latest version and solve the problem.
Also, reinstall `genesis-world` may also work.

closed 2024-12-22 by Yzhang367

neil-tan, 2025-02-11:
@Yzhang367 
Python needs to be built with Tk.
I've provided some instructions [here](https://github.com/neil-tan/genRL)