Genesis-Embodied-AI/genesis-world #161: Cloth simulation

alelordelo, 2024-12-20:
Can Genesis do cloth simulation on a walking avatar?

I found this demo for a simple cloth material, but wondering if collision avatar-> cloth would also work?

import genesis as gs

########################## init ##########################
gs.init()

########################## create a scene ##########################

scene = gs.Scene(
    sim_options=gs.options.SimOptions(
        dt       = 4e-3,
        substeps = 10,
    ),
    viewer_options=gs.options.ViewerOptions(
        camera_fov = 30,
        res        = (1280, 720),
        max_FPS    = 60,
    ),
    show_viewer = True,
)

########################## entities ##########################
plane = scene.add_entity(
    morph=gs.morphs.Plane(),
)

cloth_1 = scene.add_entity(
    material=gs.materials.PBD.Cloth(),
    morph=gs.morphs.Mesh(
        file='meshes/cloth.obj',
        scale=2.0,
        pos=(0, 0, 0.5), 
        euler=(0.0, 0, 0.0),
    ),
    surface=gs.surfaces.Default(
        color=(0.2, 0.4, 0.8, 1.0),
        vis_mode='visual',
    )
)

cloth_2 = scene.add_entity(
    material=gs.materials.PBD.Cloth(),
    morph=gs.morphs.Mesh(
        file='meshes/cloth.obj',
        scale=2.0,
        pos=(0, 0, 1.0), 
        euler=(0.0, 0, 0.0),
    ),
    surface=gs.surfaces.Default(
        color=(0.8, 0.4, 0.2, 1.0),
        vis_mode='particle',
    )
)

########################## build ##########################
scene.build()

wangyian-me, 2024-12-20:
By now, the avatar is still just a rigid articulated body. So the answer is yes. We are trying to add skinning to avatar and enable collision checking on it in future versions.

alelordelo, 2024-12-20:
Would it help if I send a skinned avatar? What file format would you need?

adding collision detection requires additional code or is it about setups?

and does cloth to cloth detection also work? Ex: a jacketed layered on top of a shirt, and pants on the bottom

zhouxian, 2024-12-22:
Avatar is actually not a very ready part (that's why we didn't even mention it in doc) and it was built for some internal projects... We will polish this part with skinning later if needed

alelordelo, 2024-12-25:
Got it, if I can help with avatar kart please let me know

closed 2025-02-23 by YilingQiao

alelordelo, 2025-06-11:
@zhouxian any news? 🙂