Genesis-Embodied-AI/genesis-world #164: Issues Importing Custom Robot Model from SolidWorks into Genesis

juanjuanjuanbing, 2024-12-20:
Thank you for your exciting work! I have encountered some issues while trying to import a custom robot model into Genesis. Here is a detailed description of the problem:

The robot was modeled in SolidWorks and exported in URDF format using a SolidWorks plugin. This URDF file was then converted to an MJCF file using the Mujoco engine. However, the resulting MJCF file fails to display in Genesis.

Troubleshooting Steps:

    Initially, I tried importing other models, such as 'go1', which displayed correctly.
    By comparing the XML file from the asset folder with the 'go1' file, I noticed that the URDF generated by SolidWorks, when exported to MJCF, has unique 'geom' tags representing the robot's body. In contrast, the XML files in the examples have two parts: 'visual' and 'collision'.
    Further investigation revealed that the 'contype' and 'conaffinity' attributes determine whether the model is displayed and whether collision detection is enabled.

Proposed Solution:
I copied all the 'geom' tags and set the 'contype' and 'conaffinity' attributes of the copied 'geom' to 0. This allowed the model to display correctly.
[image]

Is my solution appropriate? Have others encountered similar issues?

Thank you for your time and assistance.

YunSang123, 2025-01-05:
Hi, I am a beginner in Genesis. Can you tell me how you imported your custom urdf file?

juanjuanjuanbing, 2025-01-06:
> Hi, I am a beginner in Genesis. Can you tell me how you imported your custom urdf file?

You  can use a plug-in（https://github.com/Genesis-Embodied-AI/Genesis/issues/164#issuecomment-2571686481）on solidworks to transfer your .stl files into .urdf
Then it can be transfered into .mjcf file by MuJoCo 
`    ~/mujoco/mujoco210/bin
    ./compile /path/to/model.urdf /path/to/model.xml`

zhouxian, 2025-01-06:
Urdf should be able to load into Genesis directly without converting to MJCF :)

juanjuanjuanbing, 2025-01-06:
> Urdf should be able to load into Genesis directly without converting to MJCF :)

You are right, transferring to the mjcf file is the route previously trained on mujoco. Thank you for your suggestion

closed 2025-02-23 by YilingQiao