Genesis-Embodied-AI/genesis-world #165: OpenGL/GLFW Conflict Between Taichi and PyGEL3D on macOS (Apple Silicon)

analogjedi, 2024-12-20:
## Environment
- macOS with Apple Silicon (M3 Max)
- Python 3.11
- Genesis Physics 0.2.0
- PyGEL3D 0.5.2
- Taichi 1.7.2

## Issue
When running the macOS example script with visualization enabled, there's a conflict between GLFW implementations from Taichi and PyGEL3D packages. This leads to OpenGL initialization issues and ultimately causes a framebuffer binding error.

The specific error starts with duplicate GLFW class implementations:
```objc
Class GLFWHelper is implemented in both /taichi/_lib/core/taichi_python.cpython-311-darwin.so and /pygel3d/libPyGEL.dylib
Class GLFWApplicationDelegate is implemented in both locations
[etc...]

This is followed by shader uniform binding failures and eventually crashes with:

```
OpenGL.error.GLError: GLError(
err = 1282,
description = b'invalid operation',
baseOperation = glBindFramebuffer,
cArguments = (GL_DRAW_FRAMEBUFFER, 0)
)
```

Steps to Reproduce

Create a Python virtual environment
Install genesis-world and its dependencies
Run the macOS example script form the /examples directory with visualization enabled (python getting_started_mac.py -v)

Expected Behavior
The visualization window should open and show the physics simulation.

Actual Behavior
The script crashes with OpenGL errors due to conflicting GLFW implementations.

DuinoDu, 2024-12-20:
I find that this bug is not caused by OpenGL/GLFW conflict. I compile PyGEL3D without OpenGL/GLFW, by setting [Use_GLGraphics to OFF in CMakeLists.txt](https://github.com/janba/GEL/blob/master/CMakeLists.txt#L25). Conflict warnings disappear but bug still exists. 
Here is a screen record for log.

https://github.com/user-attachments/assets/68b3bfaf-dfe2-43b3-b511-45414e0787bf

DuinoDu, 2024-12-20:
I fixed it by replacing `genesis/ext/pyrender` with [pyrender offical one](https://github.com/mmatl/pyrender), with some update compatible with genesis. Code is here https://github.com/DuinoDu/Genesis. Plan to push a pull request after fixing other compatible bugs.

https://github.com/user-attachments/assets/c0ea8d08-114f-4fc9-93ab-d8d1ecec7963

arch1baald, 2024-12-20:
The same issue on Python 3.10 and 3.12

VioletBenin, 2024-12-20:
I solve this problem by setting the older version of PyGEL3D
#177

arch1baald, 2024-12-20:
> I solve this problem by setting the older version of "PyGEL3D == 0.3.1", #177
This fix doesn't work for me.

```
  File "/Users/arch1baald/proj/Genesis/.venv/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/Users/arch1baald/proj/Genesis/.venv/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 654, in on_draw
    self._renderer.render_texts(
  File "/Users/arch1baald/proj/Genesis/.venv/lib/python3.11/site-packages/genesis/ext/pyrender/renderer.py", line 258, in render_texts
    self._configure_forward_pass_viewport(0)
  File "/Users/arch1baald/proj/Genesis/.venv/lib/python3.11/site-packages/genesis/ext/pyrender/renderer.py", line 1160, in _configure_forward_pass_viewport
    glBindFramebuffer(GL_DRAW_FRAMEBUFFER, 0)
  File "/Users/arch1baald/proj/Genesis/.venv/lib/python3.11/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
	err = 1282,
	description = b'invalid operation',
	baseOperation = glBindFramebuffer,
	cArguments = (GL_DRAW_FRAMEBUFFER, 0)
)
```

```
> uname -a
Darwin <> 23.6.0 Darwin Kernel Version 23.6.0: Mon Jul 29 21:13:04 PDT 2024; root:xnu-10063.141.2~1/RELEASE_ARM64_T6020 arm64

> pip freeze
absl-py==2.1.0
annotated-types==0.7.0
certifi==2024.12.14
charset-normalizer==3.4.0
coacd==1.0.5
colorama==0.4.6
contourpy==1.3.1
cycler==0.12.1
Cython==3.0.11
dataclasses-json==0.6.7
decorator==5.1.1
Deprecated==1.2.15
dill==0.3.9
etils==1.11.0
filelock==3.16.1
fonttools==4.55.3
freetype-py==2.5.1
fsspec==2024.12.0
genesis-world==0.2.0
glfw==2.8.0
idna==3.10
imageio==2.36.1
imageio-ffmpeg==0.5.1
importlib_resources==6.4.5
Jinja2==3.1.4
kiwisolver==1.4.7
lazy_loader==0.4
libigl==2.5.1
llvmlite==0.43.0
lxml==5.3.0
markdown-it-py==3.0.0
MarkupSafe==3.0.2
marshmallow==3.23.2
matplotlib==3.10.0
mdurl==0.1.2
moviepy==2.1.1
mpmath==1.3.0
mujoco==3.2.5
mypy-extensions==1.0.0
networkx==3.4.2
numba==0.60.0
numpy==1.26.4
opencv-python==4.10.0.84
OpenEXR==3.3.2
packaging==24.2
pillow==10.4.0
platformdirs==4.3.6
plotly==5.24.1
pooch==1.8.2
proglog==0.1.10
psutil==6.1.1
pycollada==0.8
pydantic==2.7.1
pydantic_core==2.18.2
PyGEL3D==0.5.2
pyglet==2.0.20
pygltflib==1.16.0
Pygments==2.18.0
pymeshlab==2023.12.post2
pyobjc-core==10.3.2
pyobjc-framework-Cocoa==10.3.2
PyOpenGL==3.1.7
pyparsing==3.2.0
python-dateutil==2.9.0.post0
python-dotenv==1.0.1
pyvista==0.44.2
requests==2.32.3
rich==13.9.4
scikit-image==0.25.0
scipy==1.14.1
scooby==0.10.0
screeninfo==0.8.1
six==1.17.0
sympy==1.13.1
taichi==1.7.2
tenacity==9.0.0
tetgen==0.6.4
tifffile==2024.12.12
torch==2.5.1
torchaudio==2.5.1
torchvision==0.20.1
tqdm==4.67.1
typing-inspect==0.9.0
typing_extensions==4.12.2
urllib3==2.2.3
vtk==9.3.1
wrapt==1.17.0
zipp==3.21.0
```

VioletBenin, 2024-12-20:
I solve this problem using the older "PyGEL3D == 0.1.0" version on MacOS M1 chip.  #179

analogjedi, 2024-12-22:
Thanks to the post of 
https://github.com/Genesis-Embodied-AI/Genesis/issues/207#issuecomment-2558123904

I was able to get the initial getting started script working on my Macbook Pro M3 Max w/ Sonoma 14.5.

I could not get it to work with my python venv's but I had success using miniconda with Python 3.10.

closed 2024-12-22 by analogjedi

duburcqa, 2025-03-12:
The crash is unrelated to Taichi dylib conflict. The actual bug has been fixed on main branch.