Genesis-Embodied-AI/genesis-world #168: Genesis fails on Apple Silicon (MPS) due to attempt to use float64

ZeroSaturn, 2024-12-20:
'''''(.venv) (base) kameron@Kamerons-MacBook-Pro Genesis % python examples/differentiable_push.py -v
objc[64494]: Class GLFWHelper is implemented in both /Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/taichi/_lib/core/taichi_python.cpython-310-darwin.so (0x17baf4570) and /Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/pygel3d/libPyGEL.dylib (0x31348d7c0). One of the two will be used. Which one is undefined.
objc[64494]: Class GLFWApplicationDelegate is implemented in both /Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/taichi/_lib/core/taichi_python.cpython-310-darwin.so (0x17baf45c0) and /Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/pygel3d/libPyGEL.dylib (0x31348d810). One of the two will be used. Which one is undefined.
objc[64494]: Class GLFWWindowDelegate is implemented in both /Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/taichi/_lib/core/taichi_python.cpython-310-darwin.so (0x17baf45e8) and /Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/pygel3d/libPyGEL.dylib (0x31348d838). One of the two will be used. Which one is undefined.
objc[64494]: Class GLFWContentView is implemented in both /Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/taichi/_lib/core/taichi_python.cpython-310-darwin.so (0x17baf4638) and /Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/pygel3d/libPyGEL.dylib (0x31348d888). One of the two will be used. Which one is undefined.
objc[64494]: Class GLFWWindow is implemented in both /Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/taichi/_lib/core/taichi_python.cpython-310-darwin.so (0x17baf46b0) and /Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/pygel3d/libPyGEL.dylib (0x31348d900). One of the two will be used. Which one is undefined.
[Genesis] [10:41:44] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [10:41:44] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [10:41:44] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [10:41:44] [INFO] Running on [Apple M1 Pro] with backend gs.metal. Device memory: 16.00 GB.
[Genesis] [10:41:44] [DEBUG] [Taichi] version 1.7.2, llvm 15.0.7, commit 0131dce9, osx, python 3.10.2
[Genesis] [10:41:44] [DEBUG] [Taichi] Starting on arch=metal
[Genesis] [10:41:44] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: 0, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [10:41:45] [INFO] Scene <b9359cc> created.
[Genesis] [10:41:45] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <7f26f4c>, morph: <gs.morphs.URDF(file='/Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/genesis/assets/urdf/plane/plane.urdf')>, material: <gs.materials.Rigid>.
[Genesis] [10:41:45] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 0.
[Genesis] [10:41:45] [INFO] Adding <gs.ToolEntity>. idx: 1, uid: <64cfd65>, morph: <gs.morphs.Mesh(file='/Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/genesis/assets/meshes/stirrer.obj')>, material: <gs.materials.Tool>.
[Genesis] [10:41:45] [DEBUG] Processing mesh: /Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/genesis/assets/meshes/stirrer.obj.
[Genesis] [10:41:46] [INFO] Adding <gs.MPMEntity>. idx: 2, uid: <3c7b5a0>, morph: <gs.morphs.Box>, material: <gs.materials.MPM.Elastic>.
[Genesis] [10:41:46] [INFO] Sampling particles with pbs-32 sampler and generating `.ptc` file:
[Genesis] [10:41:46] [WARNING] `pbs` sampler failed. Falling back to `random` sampler.
[Genesis] [10:41:46] [DEBUG] Sampled particles (`.ptc`) found in cache.
Traceback (most recent call last):
  File "/Users/kameron/PycharmProjects/GenesisEngine/Genesis/examples/differentiable_push.py", line 152, in <module>
    main()
  File "/Users/kameron/PycharmProjects/GenesisEngine/Genesis/examples/differentiable_push.py", line 49, in main
    obj1 = scene.add_entity(
  File "/Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/genesis/engine/scene.py", line 351, in add_entity
    entity = self._sim._add_entity(morph, material, surface, visualize_contact)
  File "/Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/genesis/engine/simulator.py", line 137, in _add_entity
    entity = self.mpm_solver.add_entity(self.n_entities, material, morph, surface)
  File "/Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/genesis/engine/solvers/mpm_solver.py", line 212, in add_entity
    entity = MPMEntity(
  File "/Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/genesis/engine/entities/mpm_entity.py", line 26, in __init__
    super().__init__(
  File "/Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/genesis/engine/entities/particle_entity.py", line 69, in __init__
    self.sample()
  File "/Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/genesis/engine/entities/particle_entity.py", line 259, in sample
    self._init_particles_offset = (gs.tensor(particles).contiguous() - gs.tensor(origin)).contiguous()
  File "<string>", line 3, in tensor
  File "/Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/genesis/grad/creation_ops.py", line 64, in torch_op_wrapper
    gs_tensor = from_torch(torch_tensor, dtype, requires_grad, detach=True, scene=scene)
  File "/Users/kameron/PycharmProjects/GenesisEngine/.venv/lib/python3.10/site-packages/genesis/grad/creation_ops.py", line 92, in from_torch
    gs_tensor = Tensor(torch_tensor.to(gs.device).to(dtype), scene=scene).clone()
TypeError: Cannot convert a MPS Tensor to float64 dtype as the MPS framework doesn't support float64. Please use float32 instead.
[Genesis] [10:41:46] [INFO] 💤 Exiting Genesis and caching compiled kernels...
(.venv) (base) kameron@Kamerons-MacBook-Pro Genesis % 

'''''

The error stems from a data type incompatibility when running on Apple Silicon Macs with the Metal Performance Shaders (MPS) backend. MPS, designed for GPU acceleration, does not support 64-bit floating-point numbers (float64). The Genesis code, or a dependent library, attempts to create or utilize a tensor with float64 precision. This triggers the error because MPS cannot handle this data type natively. The core issue is a mismatch between the code's default or implicit use of float64 and the hardware's limitations.

wangyian-me, 2024-12-20:
Since it is based mostly on Taichi, you can actually find the supported type system right [here](https://docs.taichi-lang.org/docs/type).

WuZhuoran, 2024-12-20:
Same Error here.

FractalCatFX, 2024-12-21:
Same, Macbook Pro M1 Max, 32 GB RAM

wangyian-me, 2024-12-22:
This [pull request](https://github.com/Genesis-Embodied-AI/Genesis/pull/8#issuecomment-2553197504) should have solved the problem.

michaltakac, 2024-12-22:
Changing this line https://github.com/Genesis-Embodied-AI/Genesis/blob/main/genesis/grad/creation_ops.py#L92 from:

```python
gs_tensor = Tensor(torch_tensor.to(gs.device).to(dtype), scene=scene).clone()
```

to

```python
gs_tensor = Tensor(torch_tensor.to(dtype).to(gs.device), scene=scene).clone()
```

makes the error "TypeError: Cannot convert a MPS Tensor to float64 dtype as the MPS framework doesn't support float64" go away. 

Although, then I encountered this problem:

```
[Genesis] [22:34:38] [INFO] Compiling simulation kernels...
RHI Error: cannot create compute pipeline state: Compiler encountered an internal error (code=3)
RHI Error: cannot create compute pipeline state: Compiler encountered an internal error (code=3)
Assertion failed: (p != nullptr), function bind_pipeline, file metal_device.mm, line 382.
```

closed 2024-12-27 by ziyanx02