Genesis-Embodied-AI/genesis-world #172: Segmentation fault on VMWare ubuntu 20.04

wwdok, 2024-12-20:
I run the [visualization example](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html), but got Segmentation fault :
```
[Genesis] [09:03:36] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [09:03:36] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [09:03:36] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [09:03:36] [INFO] Running on [Intel(R) Core(TM) i7-8700 CPU @ 3.20GHz] with backend gs.cpu. Device memory: 3.79 GB.
[Genesis] [09:03:36] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [09:03:37] [INFO] Scene <c5a3c32> created.
[Genesis] [09:03:37] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <95c3f60>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [09:03:37] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <ccf9353>, morph: <gs.morphs.MJCF(file='/home/user/miniconda3/lib/python3.12/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [09:03:38] [INFO] Building scene <c5a3c32>...
[Genesis] [09:03:43] [INFO] Compiling simulation kernels...
[Genesis] [09:03:49] [INFO] Building visualizer...
Segmentation fault```

segrey2k, 2024-12-26:
python examples/locomotion/go2_eval.py
[Genesis] [07:16:11] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [07:16:11] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [07:16:11] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [07:16:11] [INFO] Running on [NVIDIA GeForce RTX 4060 Ti] with backend gs.cuda. Device memory: 15.61 GB.
[Genesis] [07:16:11] [INFO] Genesis initialized. version: 0.2.0, seed: None, precision: '32',debug: False, theme: 'dark'.
[Genesis] [07:16:11] [INFO] Scene <e5296ac> created.
[Genesis] [07:16:11] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <7da6fc0>, morph: <gs.morphs.URDF(file='/home/user0/myenv/lib/python3.11/site-packages/genesis/assets/urdf/plane/plane.urdf')>, material: <gs.materials.Rigid>.
[Genesis] [07:16:11] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <e5c46d4>, morph: <gs.morphs.URDF(file='/home/user0/myenv/lib/python3.11/site-packages/genesis/assets/urdf/go2/urdf/go2.urdf')>, material: <gs.materials.Rigid>.
[Genesis] [07:16:13] [INFO] Building scene <e5296ac>...
[Genesis] [07:16:18] [INFO] Compiling simulation kernels...
[Genesis] [07:16:25] [INFO] Building visualizer...
Segmentation fault

Same problems.
I work via RDP on ubuntu xrdp server from Windows PC.
I can work with Genesis with "show_viewer = False" in scene and see writing video from camera only.
In examples I make these changes, but I can't make its in Go2Env object.
Any ideas?

segrey2k, 2024-12-26:
With debug logging level:

./ClothPBD_3.py

[Genesis] [08:26:59] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [08:26:59] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [08:26:59] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [08:26:59] [INFO] Running on [NVIDIA GeForce RTX 4060 Ti] with backend gs.cuda. Device memory: 15.61 GB.
[Genesis] [08:26:59] [DEBUG] [Taichi] version 1.7.2, llvm 15.0.4, commit 0131dce9, linux, python 3.11.2
[Genesis] [08:26:59] [DEBUG] [Taichi] Starting on arch=cuda
[Genesis] [08:26:59] [INFO] Genesis initialized. version: 0.2.0, seed: None, precision: '32',debug: False, theme: 'dark'
[Genesis] [08:26:59] [INFO] Scene <0b784ee> created.
[Genesis] [08:26:59] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <f4e9fac>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [08:26:59] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 0.
[Genesis] [08:26:59] [INFO] Adding <gs.PBD2DEntity>. idx: 1, uid: <f288c3f>, morph: <gs.morphs.Mesh(file='/home/user0/myenv/lib/python3.11/site-packages/genesis/assets/meshes/cloth.obj')>, material: <gs.PBD.Cloth>.
[Genesis] [08:26:59] [DEBUG] Remeshed file (`.rm`) found in cache.
[Genesis] [08:26:59] [INFO] Adding <gs.PBD2DEntity>. idx: 2, uid: <947b951>, morph: <gs.morphs.Mesh(file='/home/user0/myenv/lib/python3.11/site-packages/genesis/assets/meshes/cloth.obj')>, material: <gs.PBD.Cloth>.
[Genesis] [08:26:59] [DEBUG] Remeshed file (`.rm`) found in cache.
[Genesis] [08:26:59] [INFO] Adding <gs.MPMEntity>. idx: 3, uid: <42bf21c>, morph: <gs.morphs.Sphere>, material: <gs.materials.MPM.Elastic>.
[Genesis] [08:26:59] [DEBUG] Sampled particles (`.ptc`) found in cache.
[Genesis] [08:26:59] [INFO] Sampled 4,159 particles.
[Genesis] [08:26:59] [INFO] Building scene <0b784ee>...
[Genesis] [08:27:05] [INFO] Compiling simulation kernels...
[Genesis] [08:27:16] [INFO] Building visualizer...
Segmentation fault

duburcqa, 2025-03-22:
Closing because of inactivity.

closed 2025-03-22 by duburcqa