Genesis-Embodied-AI/genesis-world #181: Simulation Speed Benchmark likely has significant issues leading to overstated numbers

StoneT2000, 2024-12-21:
Just some important context, I have went through the Genesis code base and benchmarks and found some issues that likely lead to inflated benchmark speeds.

I've written a full report on it here: https://x.com/stone_tao/status/1870243004730225009?s=46&t=LBFTca4dqDdDCjhzaM56tA

The main issue in question is why the original speed benchmark 
1. uses the fastest physics setting when other tutorial/example code don't
2. takes one action followed by 999 steps of no actions (which likely means the rigid body solver is early exiting or something most of the time leading to faster speed?)
3. disables robot self collisions

[image]

Sharing it here so people can see and verify numbers themselves on at least the Genesis only benchmark. I look forward to improvements that address these issues! Genesis still stands as an impressive effort to integrate many physics solvers into one package.

ziyanx02, 2024-12-21:
Thank you for bringing this up. To show the results for both benchmarking tasks, here is the robot dog result extracted from the comment in your code.

|Original|*+ more sim accuracy*|+ self collision|+ random action|
|--|--|--|--|
|14.5M|8M|5.5M-7.5M|1.2M|\

Allow me to provide some explanation within the context of the locomotion. 
1. *+ more sim accuracy* here is simply setting substeps=2. However, for locomotion tasks such as backflip, we use dt=0.005, substep=1,  decimation=4, which are widely used in other locomotion works using IsaacGym. We successfully deployed the first double backflip policy obtained in the simulation, so there is no fidelity issue (at least) in locomotion setting. From my perspective of view, setting substeps=2 only causes the speed test results to decrease by half. 
2. In the locomotion example, we use substeps=2 because there is no need to introduce decimation because control_dofs_pos in Genesis will update the applied torques in each substep.

By the way, here are my results from a 6000Ada:

|Original|substeps=2|+ self collision|+ random action|
|--|--|--|--|
|16.4M|-|15.9M|6M|
|16.4M|8M|7.8M|3.4M|

It is notable that the decreasing proportion varies significantly, possibly due to differences in system environments. Therefore, it's too early to draw a conclusion at this point.

StoneT2000, 2024-12-21:
Interesting that it runs faster on 6000Ada. But I don't know enough about Taichi to understand how it leverages GPUs compared to jax based systems or PhysX. I will just discuss 4090 results since this is what is being stated and claimed on the project website and online posts.

Whether the sim2real transfer of the double backflip policy is reported correctly I leave it up to the community to decide and agree that part at least is too early to conclude. If it works then that's a great contribution although I'm sure the same setup would work in MJX/Isaac. I'm not an expert in locomotion but I have been told that without released benchmark code of MJX/Isaac and details about training it's hard to say whether Genesis's original reported numbers are relatively accurate.

Also just to clarify terms, decimation and substeps here refers to what exactly? I do not see decimation in this codebase, only substeps. Looking at the code it sounds like `substeps` is the standard `decimation` term as I see in the `scene.step` function call it calls `sim.step` which loops `substeps` time over `rigid_solver.substep()`. Or is `decimation` the number of calls to `scene.step` before passing observations to a neural network and generating actions to send to the simulator? Understanding this will better help me understand how one should benchmark FPS during RL correctly for Genesis.

Finally, your explanations do not address the figure I shared in my first comment, showing the issues with the Franka arm simulation speeds where it goes from 43M down to < 1M which is one of the two benchmarking tasks (and the one I'm concerned the most about). Those numbers corresponding directly to the claim made in Zhou Xian's original tweet/Genesis's website that Genesis runs at 43 million FPS which I find hard to believe is realistic. Regarding the substeps argument for the benchmark I already am being conservative by running the franka arm move benchmark with `substeps=2` instead of `substeps=4` which is recommended by the Genesis documentation to get more stable manipulation/grasps.

ziyanx02, 2024-12-21:
We will respond to your question after conducting more extensive tests.

StoneT2000, 2025-01-08:
Thanks to the authors for the comprehensive report. Don't have time to verify things but the numbers in the report look more accurate when given the right context: https://placid-walkover-0cc.notion.site/genesis-performance-benchmarking?pvs=4

closed 2025-01-08 by StoneT2000