Genesis-Embodied-AI/genesis-world #184: libEGL Error

Yzhang367, 2024-12-21:
When I ran the `go2_train.py` in the rsl_rl folder, I came across the following error message.
CPU: 13th Gen Intel(R) Core(TM) i9-13980HX
GPU: GeForce RTX 4060 Laptop
Memory: 16GB
```Terminal
W 12/21/24 13:03:04.927 6923] [cuda_driver.cpp:load_lib@36] libcuda.so lib not found.
[W 12/21/24 13:03:04.927 6923] [misc.py:adaptive_arch_select@758] Arch=[<Arch.cuda: 3>] is not supported, falling back to CPU
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
libEGL warning: egl: failed to create dri2 screen
Traceback (most recent call last):
  File "/home/yzhang/桌面/rsl_rl/Genesis/examples/locomotion/go2_train.py", line 170, in <module>
    main()
  File "/home/yzhang/桌面/rsl_rl/Genesis/examples/locomotion/go2_train.py", line 155, in main
    env = Go2Env(
          ^^^^^^^
  File "/home/yzhang/桌面/rsl_rl/Genesis/examples/locomotion/go2_env.py", line 68, in __init__
    self.scene.build(n_envs=num_envs)
  File "/home/yzhang/.conda/envs/genesis/lib/python3.12/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yzhang/.conda/envs/genesis/lib/python3.12/site-packages/genesis/engine/scene.py", line 556, in build
    self._visualizer.build()
  File "/home/yzhang/.conda/envs/genesis/lib/python3.12/site-packages/genesis/vis/visualizer.py", line 110, in build
    self._rasterizer.build()
  File "/home/yzhang/.conda/envs/genesis/lib/python3.12/site-packages/genesis/vis/rasterizer.py", line 31, in build
    self._renderer = pyrender.OffscreenRenderer(
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yzhang/.conda/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/offscreen.py", line 29, in __init__
    self._create(pyopengl_platform)
  File "/home/yzhang/.conda/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/offscreen.py", line 193, in _create
    self._platform.init_context()
  File "/home/yzhang/.conda/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/platforms/egl.py", line 205, in init_context
    assert eglInitialize(self._egl_display, major, minor)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yzhang/.conda/envs/genesis/lib/python3.12/site-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/yzhang/.conda/envs/genesis/lib/python3.12/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.raw.EGL._errors.EGLError: EGLError(
        err = EGL_NOT_INITIALIZED,
        baseOperation = eglInitialize,
        cArguments = (
                <OpenGL._opaque.EGLDisplay_pointer object at 0x7de5ec34f6d0>,
                c_long(0),
                c_long(0),
        ),
        result = 0
)
```
I tried to reinstall `libnvidia-egl-wayland1` but it didn't work. And I thought the error might relate to my CUDA version or my memory limits. When I tried to install another CUDA, it conflicted with the existing one built by pytorch.

Yzhang367, 2024-12-21:
I have solved the problem by installing 12.6 version CUDA with .deb format file then reboot.

closed 2024-12-22 by Yzhang367

duburcqa, 2025-04-25:
The specific EGL device issue you described was actually fixed by #1071. Your solution worked because you avoided the fallback mechanism in Genesis entirely, which was actually broken.