Genesis-Embodied-AI/genesis-world #185: The window pops up but only displays a black screen, with no rendered content.

SuanCai-Fish, 2024-12-21:
Ubuntu 22.04  Python 3.9 RTX 4090 
Use SSH（remote desktop ）
[image]
First，there is a problem that **"pyglet.gl.ContextException: Could not create GL context "**
and I use this mothed.
`conda install -c conda-forge libstdcxx-ng `
This resolved the initial error, but a new issue has arisen. When running the program, the window successfully pops up, but the content only displays a black screen
[image]
Meanwhile, I change this:
[image]
but it does't work.

I don't know how to sovle it :(

MaxtomaxMax, 2024-12-23:
maybe you need to wait a bit long (I assume you're using X11-forwarding)

kassiokienitz, 2024-12-23:
I'm geting the same only black screem anf no errors on terminal. Runing on Ubuntu22 and a Hyper-V VM

ALS-eesoline, 2025-01-20:
> I'm geting the same only black screem anf no errors on terminal. Runing on Ubuntu22 and a Hyper-V VM. Mybe the problem is the X11 forwarding?

I get the same empty black window, with an extra warning "[frontend_ir.cpp:begin_frontend_struct_for_on_snode@1678] ti.loop_config(serialize=True) does not have effect on the struct for. The execution order is not guaranteed."

Have the problem been solved?

duburcqa, 2025-03-22:
All known and reproducible rendering issues have been fixed on main branch. Feel free to reopen if it still does not work for you.

closed 2025-03-22 by duburcqa