Genesis-Embodied-AI/genesis-world #189: ImportError: /home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/pymeshlab/lib/libmeshlab-common.so: undefined symbol: _ZdlPvm, version Qt_5

OldJohn86, 2024-12-21:
(genesis_env) john@john-MS-7D99:~/github/Genesis$ python examples/tutorials/pbd_cloth.py 
[Genesis] [16:09:25] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [16:09:25] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [16:09:25] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [16:09:25] [INFO] Running on [NVIDIA GeForce RTX 4080 SUPER] with backend gs.cuda. Device memory: 15.69 GB.
[Genesis] [16:09:25] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [16:09:25] [INFO] Scene <c43a18f> created.
[Genesis] [16:09:25] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <c31dd9b>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [16:09:25] [INFO] Adding <gs.PBD2DEntity>. idx: 1, uid: <36bdce0>, morph: <gs.morphs.Mesh(file='/home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/genesis/assets/meshes/cloth.obj')>, material: <gs.PBD.Cloth>.
[Genesis] [16:09:25] [INFO] Remeshing for tetrahedralization...
Traceback (most recent call last):
  File "/home/john/github/Genesis/examples/tutorials/pbd_cloth.py", line 26, in <module>
    cloth_1 = scene.add_entity(
  File "/home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "/home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/genesis/engine/scene.py", line 351, in add_entity
    entity = self._sim._add_entity(morph, material, surface, visualize_contact)
  File "/home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/genesis/engine/simulator.py", line 143, in _add_entity
    entity = self.pbd_solver.add_entity(self.n_entities, material, morph, surface)
  File "/home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/genesis/engine/solvers/pbd_solver.py", line 215, in add_entity
    entity = PBD2DEntity(
  File "/home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/genesis/engine/entities/pbd_entity.py", line 169, in __init__
    super().__init__(
  File "/home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/genesis/engine/entities/pbd_entity.py", line 34, in __init__
    super().__init__(
  File "/home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/genesis/engine/entities/particle_entity.py", line 69, in __init__
    self.sample()
  File "/home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/genesis/engine/entities/pbd_entity.py", line 187, in sample
    super().sample()
  File "/home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/genesis/engine/entities/pbd_entity.py", line 47, in sample
    self._mesh.remesh(edge_len_abs=self.particle_size, fix=isinstance(self, PBD3DEntity))
  File "/home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/genesis/engine/mesh.py", line 102, in remesh
    import pymeshlab
  File "/home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/pymeshlab/__init__.py", line 11, in <module>
    from .pmeshlab import *
ImportError: /home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/pymeshlab/lib/libmeshlab-common.so: undefined symbol: _ZdlPvm, version Qt_5
[Genesis] [16:09:25] [INFO] 💤 Exiting Genesis and caching compiled kernels...
(genesis_env) john@john-MS-7D99:~/github/Genesis$

OldJohn86, 2024-12-21:
env:
Ubuntu24.04 
python3.9

Thanks!

DongzhenHuangfu, 2025-01-02:
Hi, I just fix the issue by running following command before running my Python script:
`export LD_LIBRARY_PATH=/home/wyatt/anaconda3/lib/python3.11/site-packages/pymeshlab/lib:$LD_LIBRARY_PATH`

What critical is:

$LD_LIBRARY_PATH needs to come after /home/wyatt/anaconda3/lib/python3.11/site-packages/pymeshlab/lib:

for you it seems to be:
`export LD_LIBRARY_PATH=/home/john/anaconda3/envs/genesis_env/lib/python3.9/site-packages/pymeshlab/lib:$LD_LIBRARY_PATH`

closed 2025-02-23 by YilingQiao

alexis779, 2025-05-30:
On Ubuntu, Qt5 library may be incompatible with [pymeshlab](https://github.com/cnr-isti-vclab/PyMeshLab) native library. As a workaround, give precedence to the *python module* QT library instead of the *Ubuntu system* QT library.

```
SITE_PACKAGES=`pip show pymeshlab | grep Location | sed 's|Location: ||'`
PYMESHLAB_LIB=$SITE_PACKAGES/pymeshlab/lib
```

Make sure the symbol is found

```
strings $PYMESHLAB_LIB/libQt5Core.so.5 | grep _ZdlPvm
```

Finally, configure `LD_LIBRARY_PATH` to overwrite QT library path,

```
LD_LIBRARY_PATH=$PYMESHLAB_LIB PYOPENGL_PLATFORM=glx python <script.py>
```