Genesis-Embodied-AI/genesis-world #191: Does/will Genesis support PBD/XPBD methods?

cjt0313, 2024-12-21:
Greaaaaaat job!!! I've read the solver part of the project, it seems like we could costumize the physic solver? Just wondering does/will Genesis support PBD/XPBD methods?

zhouxian, 2024-12-21:
We do have a PBD solver for cloth and 3D elastic body. Is this what you are asking or there's more you want to add?

cjt0313, 2024-12-22:
> We do have a PBD solver for cloth and 3D elastic body. Is this what you are asking or there's more you want to add?

Thx for your reply, i got it. And one more thing for curiosity, I've noticed that the differentiability of IPC is not released, which quater (or month XD) do you plan to release the differentiability of rigid body? Will is support tasks like robot design?

zhouxian, 2024-12-22:
Rigid body differentiable sim is one of the few things that have highest priority currently. Currently it seems we can release it in ~2 months or so, but depending on the manpower we have. IPC is different, and should take a bit longer than diff rigid.

cjt0313, 2024-12-22:
> Rigid body differentiable sim is one of the few things that have highest priority currently. Currently it seems we can release it in ~2 months or so, but depending on the manpower we have. IPC is different, and should take a bit longer than diff rigid.

Thx

closed 2024-12-22 by cjt0313