Genesis-Embodied-AI/genesis-world #194: After installing LuisaCompute, when using the Ray Tracing Renderer, Backend 'cuda' is not installed.

Snow-Pigeon, 2024-12-21:
I encountered some errors when using `gs.renderers.RayTracer`.
```
[Genesis] [15:30:42] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [15:30:42] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [15:30:42] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [15:30:42] [INFO] Running on [NVIDIA GeForce RTX 4060 Laptop GPU] with backend gs.cuda. Device memory: 7.54 GB.
[Genesis] [15:30:42] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[2024-12-21 15:30:42.811] [console] [error] Backend 'cuda' is not installed. [/home/snow/Genesis/Genesis/genesis/ext/LuisaRender/src/compute/src/runtime/context.cpp:77]
     0 [0x7f93f03ddc0e]: /home/snow/Genesis/Genesis/genesis/ext/LuisaRender/build/bin/liblc-runtime.so :: luisa::compute::detail::ContextImpl::load_backend(std::__cxx11::basic_string<char, std::char_traits<char>, luisa::allocator<char> > const&) + 1422
     1 [0x7f93f03da351]: /home/snow/Genesis/Genesis/genesis/ext/LuisaRender/build/bin/liblc-runtime.so :: luisa::compute::Context::create_device(std::basic_string_view<char, std::char_traits<char> >, luisa::compute::DeviceConfig const*, bool) + 145
     2 [0x7f940ab9f598]: /home/snow/Genesis/Genesis/genesis/ext/LuisaRender/build/bin/LuisaRenderPy.cpython-310-x86_64-linux-gnu.so :: unknown + 189848
     3 [0x7f940abbfba2]: /home/snow/Genesis/Genesis/genesis/ext/LuisaRender/build/bin/LuisaRenderPy.cpython-310-x86_64-linux-gnu.so :: unknown + 322466
     4 [0x7f940ab92188]: /home/snow/Genesis/Genesis/genesis/ext/LuisaRender/build/bin/LuisaRenderPy.cpython-310-x86_64-linux-gnu.so :: unknown + 135560
     5 [0x55918d724282]: /bin/python3 :: unknown + 1618562
     6 [0x55918d71ab4b]: /bin/python3 :: _PyObject_MakeTpCall + 603
     7 [0x55918d7149ea]: /bin/python3 :: _PyEval_EvalFrameDefault + 26394
     8 [0x55918d719d14]: /bin/python3 :: _PyObject_FastCallDictTstate + 196
     9 [0x55918d72df64]: /bin/python3 :: unknown + 1658724
    10 [0x55918d71aaec]: /bin/python3 :: _PyObject_MakeTpCall + 508
    11 [0x55918d713f4d]: /bin/python3 :: _PyEval_EvalFrameDefault + 23677
    12 [0x55918d724aec]: /bin/python3 :: _PyFunction_Vectorcall + 124
    13 [0x55918d719dbd]: /bin/python3 :: _PyObject_FastCallDictTstate + 365
    14 [0x55918d72df64]: /bin/python3 :: unknown + 1658724
    15 [0x55918d71aaec]: /bin/python3 :: _PyObject_MakeTpCall + 508
    16 [0x55918d7149ea]: /bin/python3 :: _PyEval_EvalFrameDefault + 26394
    17 [0x55918d724aec]: /bin/python3 :: _PyFunction_Vectorcall + 124
    18 [0x55918d732882]: /bin/python3 :: PyObject_Call + 290
    19 [0x55918d710f59]: /bin/python3 :: _PyEval_EvalFrameDefault + 11401
    20 [0x55918d724aec]: /bin/python3 :: _PyFunction_Vectorcall + 124
    21 [0x55918d719dbd]: /bin/python3 :: _PyObject_FastCallDictTstate + 365
    22 [0x55918d72df64]: /bin/python3 :: unknown + 1658724
    23 [0x55918d71aaec]: /bin/python3 :: _PyObject_MakeTpCall + 508
    24 [0x55918d7149ea]: /bin/python3 :: _PyEval_EvalFrameDefault + 26394
    25 [0x55918d7f3e56]: /bin/python3 :: unknown + 2469462
    26 [0x55918d7f3d26]: /bin/python3 :: PyEval_EvalCode + 134
    27 [0x55918d81aae8]: /bin/python3 :: unknown + 2628328
    28 [0x55918d8152ef]: /bin/python3 :: unknown + 2605807
    29 [0x55918d81a885]: /bin/python3 :: unknown + 2627717
    30 [0x55918d819e68]: /bin/python3 :: _PyRun_SimpleFileObject + 424
    31 [0x55918d819b47]: /bin/python3 :: _PyRun_AnyFileObject + 71
    32 [0x55918d80e02e]: /bin/python3 :: Py_RunMain + 702
    33 [0x55918d7e7d6d]: /bin/python3 :: Py_BytesMain + 45
    34 [0x7f955b229d90]: /lib/x86_64-linux-gnu/libc.so.6 :: unknown + 171408
    35 [0x7f955b229e40]: /lib/x86_64-linux-gnu/libc.so.6 :: __libc_start_main + 128
    36 [0x55918d7e7c65]: /bin/python3 :: _start + 37
[1]    106033 IOT instruction (core dumped)  /bin/python3 /home/snow/Genesis/Genesis/Visualization.py
```
It indicates that my CUDA is not installed correctly, but when I use Rasterizer with `backend = gs.cuda`, it works fine. Also, there were no errors when compiling LuisaRenderPy with cmake.
```
[100%] Built target luisa-render-pipe-render
[100%] Built target LuisaRenderPy
```
The versions I am using are: python=3.10.12, ubuntu 22.04, nvidia-driver version: 550.120, cuda: 12.4

Xu-Xuanhui, 2024-12-21:
In my experience, maybe you shoud reinstall nvidia-driver=535, cuda=12.2

bradmcgo, 2024-12-21:
I encountered the same issue on my device.

- Fix:
  1. I removed my existing LuisaRender build: 'rm -rf build'
  2. I then rebuilt LuisaRender **using Python 3.10**: cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.10 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON -D LUISA_COMPUTE_ENABLE_GUI=OFF

demo.py is now building correctly.

ACMLCZH, 2024-12-26:
It seems that CMake is not finding the correct version of the CUDA Toolkit (which requires CUDA 12.0 or higher).
Issue #275 could be a helpful reference for a solution.

Possible reasons for this issue include:
1. **CUDA Toolkit Version**: You don't have CUDA Toolkit 12.0 or higher installed. You can download it from NVIDIA's CUDA 12.1.0 download archive.
2. **Version Conflict**: Your current CUDA installation conflicts with an older version. To resolve this, either remove the older version entirely or specify the CUDA Toolkit version with the following options:
`-D CMAKE_CUDA_COMPILER=/usr/local/cuda-12.X/bin/nvcc` or `-D CUDAToolkit_ROOT=/usr/local/cuda-12.X`

If you have correctly configured your CUDA Toolkit location, you should see a message like this during compilation:
```
-- Found CUDAToolkit: /usr/local/cuda-12.X/targets/x86_64-linux/include (found version "12.X.XX")
-- Build with CUDA backend: 12.X.XX
```

Hope this helps!

closed 2025-02-23 by YilingQiao