Genesis-Embodied-AI/genesis-world #196: how to deploy the policy to the real robot

sanbens1234, 2024-12-21:
Hi,
thanks for your great work!
I'm new to rl. I have get the model_100.pt after train. I wonder how to deploy the policy to the reall unitree go 2 robot. Thanks in advance

zhouxian, 2024-12-21:
Thank you for asking! In fact most labs have their own pipeline for real robot deployment, but we can certainly add that to our tutorial. @zhenjia-xu could you handle this?

abinayadinesh1, 2025-01-18:
this would be extremeley appreciated!

MrRobotoW, 2025-02-04:
I added joystick inputs to the training go2_train.py and in sim works well. Looking forward to your Sim2real deploy process to try out on my Go2. Thanks.

HasarinduPerera, 2025-05-09:
Hi, any update on this matter? Thank you!

zied-collab404, 2025-09-07:
Hi, are you willing to add the sim2real future in the near future? Thank you.

closed 2026-02-14 by duburcqa