Genesis-Embodied-AI/genesis-world #197: how should I fix this? I run it under windows enviroment

Jup33Q, 2024-12-21:
[image]

Jup33Q, 2024-12-21:
[Genesis] [18:18:24] [INFO] ╭───────────────────────────────────────────────╮
[Genesis] [18:18:24] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
[Genesis] [18:18:24] [INFO] ╰───────────────────────────────────────────────╯
[Genesis] [18:18:24] [INFO] Running on [NVIDIA RTX 6000 Ada Generation] with backend gs.cuda. Device memory: 47.99 GB.
[Genesis] [18:18:25] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [18:18:25] [INFO] Scene <eea2265> created.
[Genesis] [18:18:25] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <bf05d80>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [18:18:25] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <2feaea2>, morph: <gs.morphs.MJCF(file='C:\Users\lenov\miniconda3\envs\genesis_demo\Lib\site-packages\genesis\assets\xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [18:18:25] [INFO] Building scene <eea2265>...
[Genesis] [18:18:37] [INFO] Compiling simulation kernels...
[Genesis] [18:18:53] [INFO] Building visualizer...
Traceback (most recent call last):
  File "C:\Users\lenov\PycharmProjects\genesis_demo\start\demo01.py", line 11, in <module>
    scene.build()
  File "C:\Users\lenov\miniconda3\envs\genesis_demo\Lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\lenov\miniconda3\envs\genesis_demo\Lib\site-packages\genesis\engine\scene.py", line 556, in build
    self._visualizer.build()
  File "C:\Users\lenov\miniconda3\envs\genesis_demo\Lib\site-packages\genesis\vis\visualizer.py", line 101, in build
    self._context.build(self._scene)
  File "C:\Users\lenov\miniconda3\envs\genesis_demo\Lib\site-packages\genesis\vis\rasterizer_context.py", line 70, in build
    self.jit = JITRenderer(self._scene, [], [])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\lenov\miniconda3\envs\genesis_demo\Lib\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in __init__
    self.gen_func_ptr()
  File "C:\Users\lenov\miniconda3\envs\genesis_demo\Lib\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
    self.gl = GLWrapper()
              ^^^^^^^^^^^
  File "C:\Users\lenov\miniconda3\envs\genesis_demo\Lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 24, in __init__
    load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
  File "C:\Users\lenov\miniconda3\envs\genesis_demo\Lib\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 52, in load_func
    func_ptr = GL.platform.ctypesloader.buildFunction(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\lenov\miniconda3\envs\genesis_demo\Lib\site-packages\OpenGL\platform\ctypesloader.py", line 96, in buildFunction
    return functionType( (name, dll), )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: function 'glGetUniformLocation' not found

Jup33Q, 2024-12-21:
# here is my code 
import genesis as gs
from genesis import gs_backend
gs.init(backend=gs_backend.cuda)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()

rhythm-e, 2024-12-21:
I've come up with the same problem with the above same code from official tutorials, under Window 11 Environment.

IChihWang, 2024-12-21:
haven't solved yet, refer to https://github.com/Genesis-Embodied-AI/Genesis/issues/12

FannVvv, 2024-12-21:
solved with the answer :https://github.com/Genesis-Embodied-AI/Genesis/issues/12#issuecomment-2557010981

it may help u

ziyanx02, 2024-12-27:
Close as a duplicate.

closed 2024-12-27 by ziyanx02