Genesis-Embodied-AI/genesis-world #200: How to disable or change the denoiser?

Vesnica, 2024-12-21:
Hi, I want to disable the denoiser for raytracer renderer(since Optix is not supported in WSL2), but 'denoise' parameter is not exist. `gs.renderers.RayTracer`'s input parameter set is inconsistent with the documentation(https://genesis-world.readthedocs.io/en/latest/api_reference/options/renderer/raytracer.html)

And I don't know how to change the denoiser to ODIN, The documentation does not mention it. I set `LUISA_COMPUTE_DOWNLOAD_OIDN=ON` when compiling LuisaRenderer.

wangyian-me, 2024-12-22:
When I tested it, setting LUISA_COMPUTE_DOWNLOAD_OIDN=ON when compiling LuisaRenderer works to change it to OIDN. Also, to close the denoise or change the denoise parameter you can find it in the [camera parameters](https://genesis-world.readthedocs.io/en/latest/api_reference/camera/camera.html).

Vesnica, 2024-12-22:
Unfortunately, I have already made the configuration as described above, but it seems that Genesis still insists on using Optix.

I copied the entire successfully built directory into the installation path of the Genesis library (where I can see the OIDN-related files exist).
```
ls ~/venv/torch/lib/python3.10/site-packages/genesis/ext/LuisaRender/build/bin/
LuisaRenderPy.cpython-310-x86_64-linux-gnu.so      libluisa-render-sampler-paddedsobol.so
ctx_5feec1acce954df690e11b75b8239ba0               libluisa-render-sampler-pmj02bn.so
dummy                                              libluisa-render-sampler-sobol.so
libOpenImageDenoise.so                             libluisa-render-sampler-tileshared.so
libOpenImageDenoise.so.2                           libluisa-render-sampler-zsobol.so
libOpenImageDenoise.so.2.2.2                       libluisa-render-sdl.so
libOpenImageDenoise_core.so.2.2.2                  libluisa-render-shape-deformablemesh.so
libOpenImageDenoise_device_cpu.so.2.2.2            libluisa-render-shape-group.so
libOpenImageDenoise_device_cuda.so.2.2.2           libluisa-render-shape-inlinemesh.so
libOpenImageDenoise_device_hip.so.2.2.2            libluisa-render-shape-instance.so
libOpenImageDenoise_device_sycl.so.2.2.2           libluisa-render-shape-loopsubdiv.so
libcudadevrt.a                                     libluisa-render-shape-mesh.so
libembree4.so                                      libluisa-render-shape-plane.so
libembree4.so.4                                    libluisa-render-shape-sphere.so
libimf.so                                          libluisa-render-shape-spheregroup.so
libintlc.so                                        libluisa-render-spectrum-hero.so
libintlc.so.5                                      libluisa-render-spectrum-srgb.so
libirng.so                                         libluisa-render-subsurface-null.so
liblc-api.so                                       libluisa-render-subsurface-uniform.so
liblc-ast.so                                       libluisa-render-surface-disney.so
liblc-backend-cpu.so                               libluisa-render-surface-glass.so
liblc-backend-cuda.so                              libluisa-render-surface-layered.so
liblc-backend-remote.so                            libluisa-render-surface-matte.so
liblc-core.so                                      libluisa-render-surface-metal.so
liblc-dsl.so                                       libluisa-render-surface-mirror.so
liblc-ext-eastl.so                                 libluisa-render-surface-mix.so
liblc-ext-stb.so                                   libluisa-render-surface-null.so
liblc-ir-v2.so                                     libluisa-render-surface-plastic.so
liblc-ir.so                                        libluisa-render-surface-substrate.so
liblc-osl.so                                       libluisa-render-texture-bumptonormal.so
liblc-runtime.so                                   libluisa-render-texture-checkerboard.so
liblc-tensor.so                                    libluisa-render-texture-concat.so
liblc-validation-layer.so                          libluisa-render-texture-constant.so
liblc-vstl.so                                      libluisa-render-texture-image.so
libluisa-render-base.so                            libluisa-render-texture-mix.so
libluisa-render-camera-ortho.so                    libluisa-render-texture-multiply.so
libluisa-render-camera-pinhole.so                  libluisa-render-texture-nishitasky.so
libluisa-render-camera-thinlens.so                 libluisa-render-texture-scale.so
libluisa-render-environment-combined.so            libluisa-render-texture-sky-precompute.so
libluisa-render-environment-directional.so         libluisa-render-texture-swizzle.so
libluisa-render-environment-grouped.so             libluisa-render-texture-uvmapping.so
libluisa-render-environment-null.so                libluisa-render-texturemapping-spherical.so
libluisa-render-environment-spherical.so           libluisa-render-texturemapping-uv.so
libluisa-render-ext-assimp.so                      libluisa-render-transform-identity.so
libluisa-render-ext-assimp.so.5                    libluisa-render-transform-lerp.so
libluisa-render-ext-assimp.so.5.2.5                libluisa-render-transform-matrix.so
libluisa-render-ext-tinyexr.so                     libluisa-render-transform-srt.so
libluisa-render-film-color.so                      libluisa-render-transform-stack.so
libluisa-render-filter-box.so                      libluisa-render-transform-view.so
libluisa-render-filter-gaussian.so                 libluisa-render-util.so
libluisa-render-filter-lanczossinc.so              libluisa_compute_api_types.a
libluisa-render-filter-mitchell.so                 libluisa_compute_backend_impl.so
libluisa-render-filter-triangle.so                 libluisa_compute_ir_static.a
libluisa-render-integrator-aov.so                  libpi_level_zero.so
libluisa-render-integrator-direct.so               libsvml.so
libluisa-render-integrator-gradientpath.so         libsycl.so.7
libluisa-render-integrator-group.so                libsycl.so.7.1.0-8
libluisa-render-integrator-megapath.so             libtbb.so
libluisa-render-integrator-megapm.so               libtbb.so.12
libluisa-render-integrator-megavpt.so              libtbb.so.12.11
libluisa-render-integrator-megavptnaive.so         libtbbbind.so.3
libluisa-render-integrator-megawave.so             libtbbbind.so.3.11
[... cut ...]
    30 [0x5593ee68be68]: python :: _PyRun_SimpleFileObject + 424
    31 [0x5593ee68bb47]: python :: _PyRun_AnyFileObject + 71
    32 [0x5593ee68002e]: python :: Py_RunMain + 702
    33 [0x5593ee659d6d]: python :: Py_BytesMain + 45
    34 [0x7fe881837d90]: /lib/x86_64-linux-gnu/libc.so.6 :: unknown + 171408
    35 [0x7fe881837e40]: /lib/x86_64-linux-gnu/libc.so.6 :: __libc_start_main + 128
    36 [0x5593ee659c65]: python :: _start + 37
Aborted (core dumped)
```
Are there any additional code adjustments required?

ACMLCZH, 2024-12-26:
If you correctly enable `LUISA_COMPUTE_DOWNLOAD_OIDN` as `ON` in the compile configuration, you should see the installation process of the OIDN denoiser, which includes messages like:
```
-- [download 99% complete]
-- [download 100% complete]
-- Downloading... done
-- extracting...
     src='path/to/Genesis/genesis/ext/LuisaRender/build/_deps/openimagedenoise-subbuild/openimagedenoise-populate-prefix/src/oidn-2.2.2.x86_64.linux.tar.gz
     dst='path/to/Genesis/genesis/ext/LuisaRender/build/_deps/openimagedenoise-src'
...
```
After this, you will find the extracted folders `openimagedenoise-src`, `openimagedenoise-build`, and `openimagedenoise-subbuild` in your `build/_deps` directory. Running the build command should then work correctly.

If you do not see these dependencies installed, you might need to remove your `build` folder and recompile.

closed 2025-02-23 by YilingQiao