Genesis-Embodied-AI/genesis-world #203: Why is it running so slow, so stuck?

SuanCai-Fish, 2024-12-21:
I tested it twice, and almost every time I had to wait nearly 15 minutes to get a normal frame rate. It's the same whether I use gs.cuda or gs.cpu. Is anyone else experiencing this? this is hello_genesis example.
[image]
[image]
[image]

zhouxian, 2024-12-21:
This seems pretty weird. What platform and hardware are you on?

zhouxian, 2024-12-21:
Also, if you are using a light terminal background, you can change the theme to 'light' in gs.init()

SuanCai-Fish, 2024-12-21:
> This seems pretty weird. What platform and hardware are you on?

H800 and Ubuntu 22.04

SuanCai-Fish, 2024-12-21:
> This seems pretty weird. What platform and hardware are you on?

Does this have anything to do with my use of a remote server? I use VNC for remote desktop connections

zhouxian, 2024-12-21:
This is indeed a bit weird and I don't have a clue yet... it's probably caused by the way you transfer the rendered image back

sudhir2016, 2024-12-21:
I was trying Hello_Genesis on Google Colab. The building visualizer step takes about 20 minutes regardless of whether I use CPU or GPU.

ziqiyang107, 2024-12-23:
I got the exactly same issue, FPS is pretty low. I'm also in a remote 4090 server docker, with ubuntu 22.04, python 3.9, cuda 12.4

renyu2016, 2024-12-24:
same issue too

yic03685, 2024-12-26:
I have the same issue but I think what @zhouxian mentioned is valid. This only happens when I enabled camera.render. It seems like it forces the rendered image to back on cpu so it slows the simulation dramatically. I have no figured out how to avoid this.

soooocold, 2024-12-26:
> I have the same issue but I think what @zhouxian mentioned is valid. This only happens when I enabled camera.render. It seems like it forces the rendered image to back on cpu so it slows the simulation dramatically. I have no figured out how to avoid this.
same error，when run `cam.render()` ，I noticed that the CPU utilization is 100% but the GPU utilization is 0，and  very slowly

soooocold, 2024-12-26:
I sloved this issue by upgrade the CUDA version to 12.6
`NVIDIA-SMI 560.35.05              Driver Version: 560.35.05      CUDA Version: 12.6
`

Taka-Hashimoto, 2024-12-28:
I encountered the same issue, but I was able to resolve it by adding the following code before importing `genesis`:

```python
import os
NVIDIA_ICD_CONFIG_PATH = '/usr/share/glvnd/egl_vendor.d/10_nvidia.json'
ICD_CONFIG_CONTENT = """{
    "file_format_version" : "1.0.0",
    "ICD" : {
        "library_path" : "libEGL_nvidia.so.0"
    }
}
"""
with open(NVIDIA_ICD_CONFIG_PATH, 'w') as f:
    f.write(ICD_CONFIG_CONTENT)
```

You can find the complete code here: [Gist link](https://gist.github.com/Taka-Hashimoto/9546241b78e6d2c85a270c39a41808ff)

formoree, 2025-01-09:
> I encountered the same issue on Google Colab, but I was able to resolve it by adding the following code before importing `genesis`:我在 Google Colab 上遇到了同样的问题，但我能够通过在导入 `genesis` 之前添加以下代码来解决它：
> 
> ```python
> import os
> NVIDIA_ICD_CONFIG_PATH = '/usr/share/glvnd/egl_vendor.d/10_nvidia.json'
> ICD_CONFIG_CONTENT = """{
>     "file_format_version" : "1.0.0",
>     "ICD" : {
>         "library_path" : "libEGL_nvidia.so.0"
>     }
> }
> """
> with open(NVIDIA_ICD_CONFIG_PATH, 'w') as f:
>     f.write(ICD_CONFIG_CONTENT)
> ```
> 
> You can find the complete code here: [Gist link](https://gist.github.com/Taka-Hashimoto/9546241b78e6d2c85a270c39a41808ff)您可以在此处找到完整的代码： [Gist 链接](https://gist.github.com/Taka-Hashimoto/9546241b78e6d2c85a270c39a41808ff)

I tried your method on my cloud server, but it doesn't seem to work.

closed 2025-05-24 by duburcqa