Genesis-Embodied-AI/genesis-world #207: Please share the versions of GPU, PyTorch, CUDA, Python, and Linux.

SuanCai-Fish, 2024-12-21:
Please share the versions of GPU, PyTorch, CUDA, Python, and Linux that allow you to run Genesis properly, so I can use them as a reference.

Habatakurikei, 2024-12-21:
Hi, I have just resolved my case.

My environment:
- Mac Studio
- Python 3.10.16 (Anaconda/conda 24.7.1)
- Sequoia 15.2
- M2 Ultra
- torch 2.5.1

Commands:
conda create --name genesis python=3.10 -y
conda activate genesis
python -m pip install genesis-world
python -m pip install torch torchvision torchaudio
python render_on_macos.py -v

Hope if it helps your case. Thanks.

Xu-Xuanhui, 2024-12-21:
Intel 14700, NVIDIA 4070,
Ubuntu 22.04 , NVIDIA535, CUDA 12.2
python 3.10
torch 2.2
It is easy to install genesis. The default version of PyOpenGL is 3.1.7, but it will encounter errors during runtime. I found that version 3.1.4 of PyOpenGL can be used. As the user guider reporting, if you followed 2.A and see “GLIBCXX_3.4.30 not found”. 
cd ~/anaconda3/envs/genesis/lib
mv libstdc++.so.6 libstdc++.so.6.old
ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6 libstdc++.so.6

 However, LuisaRender has strict requirements for Ubuntu and CUDA versions.

DuinoDu, 2024-12-21:
> Hi, I have just resolved my case.
>
> My environment:
>
> * Mac Studio
> * Python 3.10.16 (Anaconda/conda 24.7.1)
> * Sequoia 15.2
> * M2 Ultra
> * torch 2.5.1
>
> Commands: conda create --name genesis python=3.10 [...]

@Habatakurikei Hi, just curious about genesis on mac. Do you meet OpenGL.error.GLError when executing `python render_on_macos.py -v`?
```
OpenGL.error.GLError: GLError(
	err = 1282,
	description = b'invalid operation',
	baseOperation = glBindFramebuffer,
	cArguments = (GL_DRAW_FRAMEBUFFER, 0)
```

Habatakurikei, 2024-12-21:
> @Habatakurikei Hi, just curious about genesis on mac. Do you meet OpenGL.error.GLError (#165) when executing `python render_on_macos.py -v`?
>
> ```
> OpenGL.error.GLError: GLError(
> 	err = 1282, [...]

No, I did not face any error including GLError through that setup.
I attach a video how `python render_on_macos.py -v` worked on my Mac studio.

https://github.com/user-attachments/assets/0f762550-09a6-434b-9f02-823385aa0fb2

oscar-lima, 2024-12-21:
- GPU: Quadro RTX 4000
- PyTorch version: 2.5.1+cu121
- CUDA Version: 12.2
- Python version: 3.10.9 (via pyenv)
- Linux: Ubuntu 20.04

analogjedi, 2024-12-22:
> Hi, I have just resolved my case.
>
> My environment:
>
> * Mac Studio
> * Python 3.10.16 (Anaconda/conda 24.7.1)
> * Sequoia 15.2
> * M2 Ultra
> * torch 2.5.1
>
> Commands: conda create --name genesis python=3.10 [...]

@Habatakurikei , thank you for posting this! I have been struggling to make any progress with a Python virtual environment (tried both Python 12 and 11). I did not try Python 10 in the venv so I am not sure if that was the reason..... but I did use your recipe with Miniconda on my Mac M3 and it finally worked! :-)

bradmcgo, 2024-12-22:
## Environment Details

**Operating System**
- Ubuntu 22.04

**Python Environment** (inside a Conda environment)
- Python 3.9
- PyTorch 2.5.1

**GPU Setup**
- CUDA 12.2
- NVIDIA Driver 535
- Hardware: RTX 3080Ti

## Installation Steps

**CUDA Installation**
- Used CUDA 12.2.2 **runfile** installer from NVIDIA Developer site: https://developer.nvidia.com/cuda-12-2-2-download-archive?target_os=Linux&target_arch=x86_64&Distribution=Ubuntu&target_version=22.04&target_type=runfile_local 
- Target: Ubuntu 22.04, x86_64

**NVIDIA Driver Installation**
- Installed via ubuntu-drivers: https://ubuntu.com/server/docs/nvidia-drivers-installation 
- Command: `sudo ubuntu-drivers install nvidia:535`

✅ Confirmed working with LuisaRender

NicholasZiglio, 2025-01-01:
### OS
Ubuntu 24.04

### Python
python 3.9

### GPU
RTX 2070 (laptop) 
NVIDIA-SMI 550.120
Driver Version: 550.120
CUDA Version: 12.4

### Pytorch
Installed Pytorch 2.5.1 in my conda genesis_env with python=3.9
`conda install pytorch torchvision torchaudio pytorch-cuda=12.4 -c pytorch -c nvidia`

### Genesis
Installed AFTER pytorch in my conda genesis_env with python=3.9
`pip install genesis-world`

### Fix pyglet.gl.ContextException: Could not create GL context
When running:

```
import genesis as gs
gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()
```

Was getting this error:

> (genesis_env) nicholas@nicholas-laptop:~/robotics/genesis$ python __init__.py 
> [Genesis] [11:13:26] [INFO] ╭───────────────────────────────────────────────────────────────╮
> [Genesis] [11:13:26] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈│
> [Genesis] [11:13:26] [INFO] ╰───────────────────────────────────────────────────────────────╯
> [Genesis] [11:13:26] [INFO] Running on [Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz] with backend gs.cpu. Device memory: 31.20 GB.
> [Genesis] [11:13:26] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
> [Genesis] [11:13:26] [INFO] Scene <e98e2c8> created.
> [Genesis] [11:13:26] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <3a69964>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
> [Genesis] [11:13:26] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <a6b6945>, morph: <gs.morphs.MJCF(file='/home/nicholas/anaconda3/envs/genesis_env/lib/python3.9/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
> [Genesis] [11:13:27] [INFO] Building scene <e98e2c8>...
> [Genesis] [11:13:32] [INFO] Compiling simulation kernels...
> [Genesis] [11:13:39] [INFO] Building visualizer...
> MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
> failed to load driver: iris
> MESA-LOADER: failed to open iris: /usr/lib/dri/iris_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
> failed to load driver: iris
> MESA-LOADER: failed to open swrast: /usr/lib/dri/swrast_dri.so: cannot open shared object file: No such file or directory (search paths /usr/lib/x86_64-linux-gnu/dri:\$${ORIGIN}/dri:/usr/lib/dri, suffix _dri)
> Exception in thread Thread-2:
> Traceback (most recent call last):
>   File "/home/nicholas/anaconda3/envs/genesis_env/lib/python3.9/threading.py", line 980, in _bootstrap_inner
>     self.run()
>   File "/home/nicholas/anaconda3/envs/genesis_env/lib/python3.9/threading.py", line 917, in run
>     self._target(*self._args, **self._kwargs)
>   File "/home/nicholas/anaconda3/envs/genesis_env/lib/python3.9/site-packages/genesis/ext/pyrender/viewer.py", line 1126, in _init_and_start_app
>     super(Viewer, self).__init__(
>   File "/home/nicholas/anaconda3/envs/genesis_env/lib/python3.9/site-packages/pyglet/window/xlib/__init__.py", line 168, in __init__
>     super().__init__(*args, **kwargs)
>   File "/home/nicholas/anaconda3/envs/genesis_env/lib/python3.9/site-packages/pyglet/window/__init__.py", line 527, in __init__
>     context = config.create_context(gl.current_context)
[... 28 lines cut ...]
> KeyboardInterrupt
> [Genesis] [11:22:17] [INFO] 💤 Exiting Genesis and caching compiled kernels...

First I checked I had mesa-utils installed (which I didn't) with the following:
`glxinfo | grep "OpenGL"`

To install:
```
sudo apt update
sudo apt install mesa-utils
```

Then I replaced the libstdc++.so.6 in my anaconda env with a symlink to my system's (thanks chatgpt):
```
mv $CONDA_PREFIX/lib/libstdc++.so.6 $CONDA_PREFIX/lib/libstdc++.so.6.bak
ln -s /usr/lib/x86_64-linux-gnu/libstdc++.so.6 $CONDA_PREFIX/lib/libstdc++.so.6
```

### Now it works!
Also, I am **not** using this before importing genesis:
```
#import os
#os.environ['PYOPENGL_PLATFORM'] = 'glx'
```
I've seen many suggest this but it wasn't necessary for me, the main issue was with `libstdc++.so.6`

sickopickle, 2025-02-04:
Is Nvidia Driver 550 fine for Ubuntu 22.04?

sickopickle, 2025-02-04:
When I run the demo wtih

Ubuntu 22.04
Pytorch 2.5.1
Driver 535
CUDA 12.2
RTX 4070 Ti

it is very laggy at 0.70 FPS. I also got a lot of warnings when compiling with cmake with stuff like "warning: unused import: crate::* 

[image]

closed 2025-03-22 by duburcqa