Genesis-Embodied-AI/genesis-world #210: LLVM ERROR

chenzhao-py, 2024-12-21:
```
Genesis] [08:58:06] [INFO] \u256d\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u256e
[Genesis] [08:58:06] [INFO] \u2502\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508 Genesis \u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2509\u2508\u2502
[Genesis] [08:58:06] [INFO] \u2570\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u2500\u256f
[Genesis] [08:58:06] [INFO] Running on [AMD EPYC 7413 24-Core Processor] with backend gs.cpu. Device memory: 503.56 GB.
[Genesis] [08:58:06] [INFO] \U0001f680 Genesis initialized. \U0001f516 version: 0.2.0, \U0001f331 seed: None, \U0001f4cf precision: '32', \U0001f41b debug: False, \U0001f3a8 theme: 'dark'.
[Genesis] [08:58:07] [INFO] Scene <bbb6941> created.
[Genesis] [08:58:07] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <8f70808>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [08:58:07] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <ff2b9a4>, morph: <gs.morphs.MJCF(file='/home/chenzhao/mambaforge3/envs/genesis_env/lib/python3.9/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [08:58:07] [INFO] Building scene <bbb6941>...
[Genesis] [08:58:24] [INFO] Compiling simulation kernels...
[Genesis] [08:59:15] [INFO] Building visualizer...
LLVM ERROR: Cannot select: 0x72849c311870: v8i32 = X86ISD::CMPP 0x72849c380560, 0x72849c334dd0, TargetConstant:i8<4>
  0x72849c380560: v8i32 = X86ISD::BLENDV 0x72849c30ee58, 0x72849c28b4c8, 0x72849c30ca28
    0x72849c30ee58: v8i32 = X86ISD::CMPP 0x72849c33e9b0, 0x72849c2942f8, TargetConstant:i8<5>
      0x72849c33e9b0: v8i32 = fabs 0x72849c28b4c8
        0x72849c28b4c8: v8i32,ch = CopyFromReg 0x7285768968b8, Register:v8i32 %4
          0x72849c289e00: v8i32 = Register %4
      0x72849c2942f8: v8f32 = X86ISD::FMAX 0x72849c313f10, 0x72849c33c650
        0x72849c313f10: v8f32 = bitcast 0x72849c317f50
          0x72849c317f50: v8i32 = and 0x72849c3116d0, 0x72849c380e20
            0x72849c3116d0: v8i32 = bitcast 0x72849c3807a0
              0x72849c3807a0: v8f32,ch = CopyFromReg 0x7285768968b8, Register:v8f32 %2
                0x72849c33be30: v8f32 = Register %2
            0x72849c380e20: v8i32,ch = X86ISD::VBROADCAST_LOAD<(load (s32) from constant-pool)> 0x7285768968b8, 0x72849c377f98
              0x72849c377f98: i64 = X86ISD::Wrapper TargetConstantPool:i64<float 0x7FFFFFFFE0000000> 0
                0x72849c31da00: i64 = TargetConstantPool<float 0x7FFFFFFFE0000000> 0
        0x72849c33c650: v8f32 = bitcast 0x72849c385748
          0x72849c385748: v8i32 = and 0x72849c313d08, 0x72849c380e20
            0x72849c313d08: v8i32 = bitcast 0x72849c2947d8
              0x72849c2947d8: v8f32,ch = CopyFromReg 0x7285768968b8, Register:v8f32 %3
                0x72849c290a60: v8f32 = Register %3
            0x72849c380e20: v8i32,ch = X86ISD::VBROADCAST_LOAD<(load (s32) from constant-pool)> 0x7285768968b8, 0x72849c377f98
              0x72849c377f98: i64 = X86ISD::Wrapper TargetConstantPool:i64<float 0x7FFFFFFFE0000000> 0
                0x72849c31da00: i64 = TargetConstantPool<float 0x7FFFFFFFE0000000> 0
      0x72849c2906b8: i8 = TargetConstant<5>
    0x72849c28b4c8: v8i32,ch = CopyFromReg 0x7285768968b8, Register:v8i32 %4
      0x72849c289e00: v8i32 = Register %4
    0x72849c30ca28: v8f32 = X86ISD::BLENDV 0x72849c28f7e0, 0x72849c3807a0, 0x72849c2947d8
      0x72849c28f7e0: v8i32 = bitcast 0x72849c333910
        0x72849c333910: v8f32 = X86ISD::CMPP 0x72849c313f10, 0x72849c33c650, TargetConstant:i8<6>
          0x72849c313f10: v8f32 = bitcast 0x72849c317f50
            0x72849c317f50: v8i32 = and 0x72849c3116d0, 0x72849c380e20
              0x72849c3116d0: v8i32 = bitcast 0x72849c3807a0
                0x72849c3807a0: v8f32,ch = CopyFromReg 0x7285768968b8, Register:v8f32 %2
                  0x72849c33be30: v8f32 = Register %2
              0x72849c380e20: v8i32,ch = X86ISD::VBROADCAST_LOAD<(load (s32) from constant-pool)> 0x7285768968b8, 0x72849c377f98
                0x72849c377f98: i64 = X86ISD::Wrapper TargetConstantPool:i64<float 0x7FFFFFFFE0000000> 0
                  0x72849c31da00: i64 = TargetConstantPool<float 0x7FFFFFFFE0000000> 0
          0x72849c33c650: v8f32 = bitcast 0x72849c385748
            0x72849c385748: v8i32 = and 0x72849c313d08, 0x72849c380e20
              0x72849c313d08: v8i32 = bitcast 0x72849c2947d8
                0x72849c2947d8: v8f32,ch = CopyFromReg 0x7285768968b8, Register:v8f32 %3
                  0x72849c290a60: v8f32 = Register %3
              0x72849c380e20: v8i32,ch = X86ISD::VBROADCAST_LOAD<(load (s32) from constant-pool)> 0x7285768968b8, 0x72849c377f98
                0x72849c377f98: i64 = X86ISD::Wrapper TargetConstantPool:i64<float 0x7FFFFFFFE0000000> 0
                  0x72849c31da00: i64 = TargetConstantPool<float 0x7FFFFFFFE0000000> 0
          0x72849c30af38: i8 = TargetConstant<6>
      0x72849c3807a0: v8f32,ch = CopyFromReg 0x7285768968b8, Register:v8f32 %2
        0x72849c33be30: v8f32 = Register %2
      0x72849c2947d8: v8f32,ch = CopyFromReg 0x7285768968b8, Register:v8f32 %3
        0x72849c290a60: v8f32 = Register %3
  0x72849c334dd0: v8f32 = BUILD_VECTOR ConstantFP:f32<0.000000e+00>, ConstantFP:f32<0.000000e+00>, ConstantFP:f32<0.000000e+00>, ConstantFP:f32<0.000000e+00>, ConstantFP:f32<0.000000e+00>, ConstantFP:f32<0.000000e+00>, ConstantFP:f32<0.000000e+00>, ConstantFP:f32<0.000000e+00>
    0x72849c2902a8: f32 = ConstantFP<0.000000e+00>
    0x72849c2902a8: f32 = ConstantFP<0.000000e+00>
    0x72849c2902a8: f32 = ConstantFP<0.000000e+00>
    0x72849c2902a8: f32 = ConstantFP<0.000000e+00>
    0x72849c2902a8: f32 = ConstantFP<0.000000e+00>
    0x72849c2902a8: f32 = ConstantFP<0.000000e+00>
    0x72849c2902a8: f32 = ConstantFP<0.000000e+00>
    0x72849c2902a8: f32 = ConstantFP<0.000000e+00>
  0x72849c375dd8: i8 = TargetConstant<4>
In function: texfunc_res_0_sam_0_0
Aborted (core dumped)
```
No clue what is causing this

zhouxian, 2024-12-21:
what system and os are you on?

chenzhao-py, 2024-12-21:
> what system and os are you on?

Linux Ubuntu 22, btw without viz, it worked fine

duburcqa, 2025-03-22:
Does this issue still occur on main branch?

closed 2025-05-24 by duburcqa