Genesis-Embodied-AI/genesis-world #212: BUGs REPORT

jindadu00, 2024-12-21:
hello, I met some bugs when I tried to simulate the motion of Allegro Hand.

1. I cannot load the URDF files using code "allegro = scene.add_entity(gs.morphs.URDF(file='allegro.urdf'),)" and raised a ERROR "AttributeError: 'NoneType' object has no attribute 'text'".The URDF files is downloaded from "https://github.com/Daniella1/urdf_files_dataset/blob/main/urdf_files/drake/allegro_hand_description/urdf/allegro_hand_description_left.urdf".

2. I managed to load the MJCF which is downloaded from "https://github.com/google-deepmind/mujoco_menagerie/blob/main/wonik_allegro/README.md" but the PD control seemed strange. My code is "import genesis as gs
import numpy as np
gs.init(backend=gs.gpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())

allegro = scene.add_entity(
    gs.morphs.MJCF(
        file  = 'wonik_allegro/left_hand.xml',
        pos   = (0.0, 0.0, 0.5),
        euler = (0, 90, 0),
    ),
)

scene.build()

jnt_names = [
    'ffj0','ffj1','ffj2','ffj3',
    'mfj0','mfj1','mfj2','mfj3',
    'rfj0','rfj1','rfj2','rfj3',
    'thj0','thj1','thj2','thj3',
]
dofs_idx = [allegro.get_joint(name).dof_idx_local for name in jnt_names]
print(dofs_idx)
# set positional gains
allegro.set_dofs_kp(
    kp             = np.array([4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500]),
    dofs_idx_local = dofs_idx,
)
# set velocity gains
allegro.set_dofs_kv(
    kv             = np.array([4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500, 4500]),
    dofs_idx_local = dofs_idx,
)
# set force range for safety
allegro.set_dofs_force_range(
    lower          = np.array([-100, -100, -100, -100, -100, -100, -100, -100, -100, -100, -100, -100, -100, -100, -100, -100]),
    upper          = np.array([ 100,  100,  100,  100,  100,  100,  100,  100,  100, -100, -100, -100, -100, -100, -100, -100]),
    dofs_idx_local = dofs_idx,
)

pos0  = np.zeros(16)
pos1 = np.ones(16) * 0.1
pos2 = np.ones(16) * 0.2
pos3 = np.ones(16) * 0.3

# PD control
for i in range(1250):
    if i == 0:
        allegro.control_dofs_position(
            pos1,
            dofs_idx,
        )
    elif i == 250:
        allegro.control_dofs_position(
            pos2,
            dofs_idx,
        )
    elif i == 500:
        allegro.control_dofs_position(
            pos3,
            dofs_idx,
        )
    elif i == 750:
        # control first dof with velocity, and the rest with position
        allegro.control_dofs_position(
            pos1[1:],
            dofs_idx[1:],
        )
        allegro.control_dofs_velocity(
            pos1[:1],
            dofs_idx[:1],
        )
    elif i == 1000:
        allegro.control_dofs_force(
            pos0,
            dofs_idx,
        )
    # This is the control force computed based on the given control command
    # If using force control, it's the same as the given control command
    print('control force:', allegro.get_dofs_control_force(dofs_idx))

    # This is the actual force experienced by the dof
    print('internal force:', allegro.get_dofs_force(dofs_idx))

    scene.step()"
But there is only 9 dofs (the dof of Allegro is 16) is controled while the control force of other dofs is max/min which is like:" [Genesis] [23:30:45] [INFO] Running at 59.94 FPS.
control force: tensor([  -0.8241,   -0.8601,   -0.8363,   -0.8277,   -0.8241,   -0.8601,
          -0.8363,   -0.8277,   -0.8274, -100.0000, -100.0000, -100.0000,
        -100.0000, -100.0000, -100.0000, -100.0000], device='cuda:0')
internal force: tensor([  -0.8359,   -0.8359,   -0.8359,   -0.8359,   -0.8359,   -0.8359,
          -0.8359,   -0.8359,   -0.8355,  -99.1898,  -99.0691,  -99.0581,
        -100.0113,  -99.0642,  -99.2019,  -99.0663], device='cuda:0')
[Genesis] [23:30:45] [INFO] Running at 59.94 FPS.
control force: tensor([  -0.8159,   -0.8519,   -0.8281,   -0.8194,   -0.8159,   -0.8519,
          -0.8281,   -0.8194,   -0.8194, -100.0000, -100.0000, -100.0000,
        -100.0000, -100.0000, -100.0000, -100.0000], device='cuda:0')
internal force: tensor([  -0.8275,   -0.8275,   -0.8275,   -0.8276,   -0.8275,   -0.8275,
          -0.8275,   -0.8276,   -0.8273,  -99.1892,  -99.0689,  -99.0562,
        -100.0092,  -99.0619,  -99.1997,  -99.0630], device='cuda:0') "

Sincerely looking forward to your respond.

ziyanx02, 2024-12-21:
5.
- The control force reaches the limit due to high values of kp and kv. With kp=4500, even a slight deviation from the default position (e.g., 0.03 radius or 2 degrees) results in a torque greater than 100.
- I suggest checking `dofs_idx` to verify if all joints are actuated.

jindadu00, 2024-12-22:
Thank you for your advice. I modified my code and **set the kp and kv to 0.1**:
```
import genesis as gs
import numpy as np
gs.init(backend=gs.gpu)

scene = gs.Scene(
    show_viewer = True,
    viewer_options = gs.options.ViewerOptions(
        res           = (1280, 960),
        camera_pos    = (3.5, 0.0, 2.5),
        camera_lookat = (0.0, 0.0, 0.5),
        camera_fov    = 40,
        max_FPS       = 60,
    ),
    vis_options = gs.options.VisOptions(
        show_world_frame = True,
        world_frame_size = 1.0,
        show_link_frame  = False,
        show_cameras     = False,
        plane_reflection = True,
        ambient_light    = (0.1, 0.1, 0.1),
    ),
    renderer=gs.renderers.Rasterizer(),
)
plane = scene.add_entity(gs.morphs.Plane())

allegro = scene.add_entity(
    gs.morphs.MJCF(
        file  = 'wonik_allegro/left_hand.xml',
        pos   = (0.0, 0.0, 0.5),
        euler = (0, 90, 0),
    ),
)
cam = scene.add_camera(
    res    = (640, 480),
    pos    = (3.5, 0.0, 2.5),
    lookat = (0, 0, 0.5),
    fov    = 30,
    GUI    = False,
)

scene.build()

jnt_names = [
    'ffj0','ffj1','ffj2','ffj3',
    'mfj0','mfj1','mfj2','mfj3',
    'rfj0','rfj1','rfj2','rfj3',
    'thj0','thj1','thj2','thj3',
]
dofs_idx = [allegro.get_joint(name).dof_idx_local for name in jnt_names]
print(dofs_idx)
# set positional gains
allegro.set_dofs_kp(
    kp             = np.array([0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1]),
    dofs_idx_local = dofs_idx,
)
# set velocity gains
allegro.set_dofs_kv(
    kv             = np.array([0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1, 0.1]),
    dofs_idx_local = dofs_idx,
)
# set force range for safety
allegro.set_dofs_force_range(
    lower          = np.array([-100, -100, -100, -100, -100, -100, -100, -100, -100, -100, -100, -100, -100, -100, -100, -100]),
    upper          = np.array([ 100,  100,  100,  100,  100,  100,  100,  100,  100, -100, -100, -100, -100, -100, -100, -100]),
    dofs_idx_local = dofs_idx,
)

pos0  = np.zeros(16)
pos1 = np.ones(16) * 0.01
pos2 = np.ones(16) * 0.02
pos3 = np.ones(16) * 0.03

cam.start_recording()

# PD control
for i in range(1250):
    if i == 0:
        allegro.control_dofs_position(
            pos1,
            dofs_idx,
        )
    elif i == 250:
        allegro.control_dofs_position(
            pos2,
            dofs_idx,
        )
    elif i == 500:
        allegro.control_dofs_position(
            pos3,
            dofs_idx,
        )
    elif i == 750:
        # control first dof with velocity, and the rest with position
        allegro.control_dofs_position(
            pos1[1:],
            dofs_idx[1:],
        )
        allegro.control_dofs_velocity(
            pos1[:1],
            dofs_idx[:1],
        )
    elif i == 1000:
        allegro.control_dofs_force(
            pos0,
            dofs_idx,
        )
    # This is the control force computed based on the given control command
    # If using force control, it's the same as the given control command
    print('control force:', allegro.get_dofs_control_force(dofs_idx))

    # This is the actual force experienced by the dof
    print('internal force:', allegro.get_dofs_force(dofs_idx))

    scene.step()
    cam.render()
cam.stop_recording(save_to_filename='video.mp4', fps=60)
```
But it seems that the motions of 10th-16th dofs are still boom at beginning and **the control force is still -100 even though send the code ```allegro.control_dofs_force(pos0,dofs_idx)```**

the logs is following:
```
[Genesis] [16:30:38] [INFO] Running at 60.00 FPS.
control force: tensor([   0.,    0.,    0.,    0.,    0.,    0.,    0.,    0.,    0., -100.,
        -100., -100., -100., -100., -100., -100.], device='cuda:0')
internal force: tensor([ 1.4036e-06, -7.9537e-05, -3.1760e-05, -6.5058e-06,  4.9167e-07,
        -7.9568e-05, -3.1767e-05, -6.5055e-06,  4.6809e-07, -9.9983e+01,
        -9.9998e+01, -9.9996e+01, -1.0005e+02, -9.9990e+01, -9.9990e+01,
        -9.9999e+01], device='cuda:0')
[Genesis] [16:30:38] [INFO] Running at 60.00 FPS.
control force: tensor([   0.,    0.,    0.,    0.,    0.,    0.,    0.,    0.,    0., -100.,
        -100., -100., -100., -100., -100., -100.], device='cuda:0')
internal force: tensor([ 1.3968e-06, -7.9384e-05, -3.1701e-05, -6.4896e-06,  4.9456e-07,
        -7.9399e-05, -3.1700e-05, -6.4885e-06,  4.7340e-07, -9.9983e+01,
        -9.9998e+01, -9.9996e+01, -1.0005e+02, -9.9990e+01, -9.9990e+01,
        -9.9999e+01], device='cuda:0')
```

the recording is following:

https://github.com/user-attachments/assets/0c6f4fdc-24df-472d-bafb-90daa0418ea8

ziyanx02, 2024-12-22:
The last seven upper bound values in `set_dofs_force_range()` are set to be -100.

closed 2024-12-22 by ziyanx02