Genesis-Embodied-AI/genesis-world #217: How to load points as entities?

kyleleey, 2024-12-21:
Hi,

Is there a way to load a large number of points(spheres) into genesis as one entity? In my case these points can be set to still, mainly for collision detection with other entites. And since in my case the points number are very large (~1e5), I assume it's not possible to directly use sphere morph for each point. 

Is the current "particle entity" suitable for this case (assign many points as particles to one entity)? And how to initialize it?

Thanks for your answer in advance!!

wangyian-me, 2024-12-22:
I don't think the current released code can do this. One hacky way is probably to initialize a "particle entity" for example MPM, and then manually fix them in each timesteps.

kyleleey, 2024-12-22:
Thanks for the reply!

In this case, is there an example code of how I can use a bunch of particle locations to initialize the morphs for this entity? Or even a hacky way of where to directly reset all the sampled particles in a MPM entity

kyleleey, 2024-12-22:
No worry I find a simple (and hacky) way is: 

1. define a new `morph` class to load particle positions, 
2. assign the material as PBD.Particle, since now only pbd_solver supports fix particle functionality,
3. add a few lines in `ParticleEntity() ` to directly use the initialized particles as the particles for simulation.

closed 2024-12-23 by kyleleey