Genesis-Embodied-AI/genesis-world #219: Unable to initialize an OpenGL 3+ context

gulj9, 2024-12-21:
runtime error while running hello_genesis.py:

[Genesis] [21:53:58] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [21:53:58] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [21:53:58] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [21:53:58] [INFO] Running on [11th Gen Intel(R) Core(TM) i9-11950H @ 2.60GHz] with backend gs.cpu. Device memory: 31.06 GB.
[Genesis] [21:53:58] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [21:53:58] [INFO] Scene <a0cf7af> created.
[Genesis] [21:53:58] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <51f6c52>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [21:53:58] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <6ab75b7>, morph: <gs.morphs.MJCF(file='/home/gentex/miniconda3/envs/g5/lib/python3.10/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [21:53:59] [INFO] Building scene <a0cf7af>...
[Genesis] [21:54:02] [INFO] Compiling simulation kernels...
[Genesis] [21:54:06] [INFO] Building visualizer...
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/home/gentex/miniconda3/envs/g5/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/home/gentex/miniconda3/envs/g5/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/gentex/miniconda3/envs/g5/lib/python3.10/site-packages/genesis/ext/pyrender/viewer.py", line 1134, in _init_and_start_app
    raise ValueError("Unable to initialize an OpenGL 3+ context")
ValueError: Unable to initialize an OpenGL 3+ context

gulj9, 2024-12-21:
-

gulj9, 2024-12-21:
-

QinghuaZH, 2024-12-22:
I hava the same question.Did you have solved the questions?

gulj9, 2024-12-22:
I haven't solve that yet but I got around it. If you disable the viewer you can dump the video to disk without using the viewer.
I found this video helpful: https://www.youtube.com/watch?v=K0mSOAXOtDs&t=283s
I used this code:

======================
import genesis as gs
import numpy as np
gs.init(backend=gs.cuda)

scene = gs.Scene(show_viewer=False)

plane = scene.add_entity(
    gs.morphs.Plane(),
)
franka = scene.add_entity(
    # gs.morphs.URDF(
    #     file='urdf/panda_bullet/panda.urdf',
    #     fixed=True,
    # ),
    gs.morphs.MJCF(file="xml/franka_emika_panda/panda.xml"),
)

cam = scene.add_camera(res=(640, 480), pos = (0,3,2), lookat=(0,0,0.5), fov=30, GUI=False)
scene.build()
cam.start_recording()

for i in range(360):
    scene.step()
    cam.set_pose(
        pos= (3.0 * np.sin(i/60), 3.0*np.cos(i/60),2),
        lookat=(0,0,0.3)
    )
    cam.render()
cam.stop_recording(save_to_filename='test1.mp4', fps=60)
=================

QinghuaZH, 2024-12-24:
But why cant be true?I am confused. Has this problem been truly resolved now?

gulj9, 2024-12-24:
I didnt find any other solution to run the viewer. It may be wrong opengl
version of some conflict between versions.

QinghuaZH, 2024-12-24:
I've solved it. I just changed the graphics card type to NVIDIA discrete graphics, and it worked. This command：sudo prime-select nvidia

gulj9, 2024-12-24:
Worked also for me! Thanks man!

**sudo prime-select nvidia**

[image]

QinghuaZH, 2024-12-24:
OK,Thanks you,too.
[image]

myuansun, 2025-01-07:
I met the same question. However when I try to run ```sudo prime-select nvidia``` it returns `Error: no integrated GPU detected.`. I am running python script in a WSL2 Ubuntu 22.04 system. `torch.cuda.is_available()` returns True with my cuda version being 12.1. Any solution to my issue?

eee336, 2025-01-10:
> I met the same question. However when I try to run `sudo prime-select nvidia` it returns `Error: no integrated GPU detected.`. I am running python script in a WSL2 Ubuntu 22.04 system. `torch.cuda.is_available()` returns True with my cuda version being 12.1. Any solution to my issue?

i also meet this problem,if you solve it，please share me ，thanks

gonzalesMK, 2025-02-12:
> I met the same question. However when I try to run ```sudo prime-select nvidia``` it returns `Error: no integrated GPU detected.`. I am running python script in a WSL2 Ubuntu 22.04 system. `torch.cuda.is_available()` returns True with my cuda version being 12.1. Any solution to my issue?

I only had a single Nvidia GPU (no integrated Intel GPU) and encountered the same issue. I was able to resolve it by running the following command:

`__NV_PRIME_RENDER_OFFLOAD=1 __GLX_VENDOR_LIBRARY_NAME=nvidia python3 tutorials/sph_liquid.py`

MikeW097, 2025-02-17:
> I met the same question. However when I try to run `sudo prime-select nvidia` it returns `Error: no integrated GPU detected.`. I am running python script in a WSL2 Ubuntu 22.04 system. `torch.cuda.is_available()` returns True with my cuda version being 12.1. Any solution to my issue?

I had issue with WSL2 as well. My understanding is the whole display chain isn't well supported on wsl, so I give up and changed to Ubuntu system instead.

closed 2025-02-23 by YilingQiao

rhl525, 2025-05-15:
> I met the same question. However when I try to run `sudo prime-select nvidia` it returns `Error: no integrated GPU detected.`. I am running python script in a WSL2 Ubuntu 22.04 system. `torch.cuda.is_available()` returns True with my cuda version being 12.1. Any solution to my issue?

I have the same problem in WSL2 :( Have you solved this problem now?