Genesis-Embodied-AI/genesis-world #221: problem running hello_genesis.py

gulj9, 2024-12-21:
Traceback (most recent call last):
  File "/home/gentex/projects/Genesis/examples/tutorials/hello_genesis.py", line 4, in <module>
    import genesis as gs
  File "/home/gentex/projects/Genesis/genesis/__init__.py", line 17, in <module>
    import torch
  File "/home/gentex/miniconda3/envs/g5/lib/python3.10/site-packages/torch/__init__.py", line 367, in <module>
    from torch._C import *  # noqa: F403
ImportError: /home/gentex/miniconda3/envs/g5/lib/python3.10/site-packages/torch/lib/../../nvidia/cusparse/lib/libcusparse.so.12: undefined symbol: __nvJitLinkComplete_12_4, version libnvJitLink.so.12

zhouxian, 2024-12-22:
it seems you failed to import torch in the first place, so it might not be a genesis problem. Could you verify that torch is supported ?

gulj9, 2024-12-22:
Thanks for your answer.
I finally succeeded to run the hello_genesis.py with using show_viewer=False
when setting show_viewer=True Im getting this error msg:

Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/home/gentex/miniconda3/envs/g6/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/home/gentex/miniconda3/envs/g6/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/gentex/projects/Genesis/genesis/ext/pyrender/viewer.py", line 1134, in _init_and_start_app
    raise ValueError("Unable to initialize an OpenGL 3+ context")
ValueError: Unable to initialize an OpenGL 3+ context

Thanks again

duburcqa, 2025-03-22:
This issue has been fixed on main branch.

closed 2025-03-22 by duburcqa