Genesis-Embodied-AI/genesis-world #224: advanced_worm.py: Cannot convert a MPS Tensor to float64 dtype as the MPS framework doesn't support float64. Please use float32 instead.

czerwinskilukasz1, 2024-12-21:
Running `Genesis/examples/tutorials/advanced_worm.py` results in an error:
```
  File "/Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/genesis/grad/creation_ops.py", line 92, in from_torch
    gs_tensor = Tensor(torch_tensor.to(gs.device).to(dtype), scene=scene).clone()
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Cannot convert a MPS Tensor to float64 dtype as the MPS framework doesn't support float64. Please use float32 instead.
```

The whole console output:

```
tutorials % python advanced_worm.py
objc[41001]: Class GLFWHelper is implemented in both /Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x12ad6b570) and /Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x16540d7d0). One of the two will be used. Which one is undefined.
objc[41001]: Class GLFWApplicationDelegate is implemented in both /Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x12ad6b5c0) and /Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x16540d820). One of the two will be used. Which one is undefined.
objc[41001]: Class GLFWWindowDelegate is implemented in both /Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x12ad6b5e8) and /Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x16540d848). One of the two will be used. Which one is undefined.
objc[41001]: Class GLFWContentView is implemented in both /Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x12ad6b638) and /Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x16540d898). One of the two will be used. Which one is undefined.
objc[41001]: Class GLFWWindow is implemented in both /Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/taichi/_lib/core/taichi_python.cpython-312-darwin.so (0x12ad6b6b0) and /Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/pygel3d/libPyGEL.dylib (0x16540d910). One of the two will be used. Which one is undefined.
[Genesis] [23:56:31] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [23:56:31] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [23:56:31] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [23:56:31] [INFO] Running on [Intel(R) Core(TM) i9-9980HK CPU @ 2.40GHz] with backend gs.metal. Device memory: 64.00 GB.
[Genesis] [23:56:31] [DEBUG] [Taichi] version 1.7.2, llvm 15.0.5, commit 0131dce9, osx, python 3.12.1
[Genesis] [23:56:31] [DEBUG] [Taichi] Starting on arch=metal
[Genesis] [23:56:31] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: 0, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [23:56:33] [INFO] Scene <e94a1c3> created.
[Genesis] [23:56:33] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <1ce5680>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [23:56:33] [DEBUG] Preprocessed `.gsd` file found in cache for geom idx 0.
[Genesis] [23:56:33] [INFO] Adding <gs.MPMEntity>. idx: 1, uid: <d1a959f>, morph: <gs.morphs.Mesh(file='/Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/genesis/assets/meshes/worm/worm.obj')>, material: <gs.materials.MPM.Muscle>.
[Genesis] [23:56:33] [INFO] Sampling particles with pbs-32 sampler and generating `.ptc` file:
[Genesis] [23:56:35] [WARNING] `pbs` sampler failed. Falling back to `random` sampler.
[Genesis] [23:56:35] [INFO] Sampling particles with random sampler and generating `.ptc` file:
Traceback (most recent call last):
  File "/Users/lukasz/Desktop/Projects/AI_Models/Genesis/examples/tutorials/advanced_worm.py", line 38, in <module>
    worm = scene.add_entity(
           ^^^^^^^^^^^^^^^^^
  File "/Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/genesis/engine/scene.py", line 351, in add_entity
    entity = self._sim._add_entity(morph, material, surface, visualize_contact)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/genesis/engine/simulator.py", line 137, in _add_entity
    entity = self.mpm_solver.add_entity(self.n_entities, material, morph, surface)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/genesis/engine/solvers/mpm_solver.py", line 212, in add_entity
    entity = MPMEntity(
             ^^^^^^^^^^
  File "/Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/genesis/engine/entities/mpm_entity.py", line 26, in __init__
    super().__init__(
  File "/Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/genesis/engine/entities/particle_entity.py", line 69, in __init__
    self.sample()
  File "/Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/genesis/engine/entities/particle_entity.py", line 259, in sample
    self._init_particles_offset = (gs.tensor(particles).contiguous() - gs.tensor(origin)).contiguous()
                                   ^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 3, in tensor
  File "/Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/genesis/grad/creation_ops.py", line 64, in torch_op_wrapper
    gs_tensor = from_torch(torch_tensor, dtype, requires_grad, detach=True, scene=scene)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/lukasz/.pyenv/versions/3.12.1/lib/python3.12/site-packages/genesis/grad/creation_ops.py", line 92, in from_torch
    gs_tensor = Tensor(torch_tensor.to(gs.device).to(dtype), scene=scene).clone()
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Cannot convert a MPS Tensor to float64 dtype as the MPS framework doesn't support float64. Please use float32 instead.
[Genesis] [23:56:35] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

Debug info:
MacBook Pro 16-inch, 2019
Processor: 2,4 GHz 8-Core Intel Core i9
macOS: Sonoma 14.6.1

wangyian-me, 2024-12-22:
My guess is that this [line](https://github.com/Genesis-Embodied-AI/Genesis/blob/da3d38ba0962b5b491ca54905c52f022ba19d92e/genesis/engine/entities/particle_entity.py#L182) should be changed to `particles.append(particles_i.astype(gs.np_float))`.

andrewlyu, 2024-12-31:
run into the same issue with example pbd_liquid.py.
Since [pull request](https://github.com/Genesis-Embodied-AI/Genesis/pull/8#issuecomment-2553197504) is supposed to have fixed this issue, should I compile the source code instead install through pip?

wangyian-me, 2025-01-01:
Yes. Compiling from source to have all the up-to-date changes.

closed 2025-01-01 by wangyian-me