Genesis-Embodied-AI/genesis-world #226: Genesis "hello world" code not functioning

arjunsshah, 2024-12-22:
After running the hello world code, I am receiving this error. I've checked everything is installed correctly in the right location.

---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[28], line 2
      1 import genesis as gs
----> 2 gs.init(backend=gs.cpu)
      4 scene = gs.Scene(show_viewer=True)
      5 plane = scene.add_entity(gs.morphs.Plane())

AttributeError: module 'genesis' has no attribute 'init'

zhouxian, 2024-12-22:
how did you install genesis?

zswang666, 2024-12-22:
You can check
```bash
ls $CONDA_PREFIX/genesis/lib/python3.10/site-packages | grep genesis # remember to change the python version accordingly
```
There shouldn't be things like `$CONDA_PREFIX/genesis/lib/python3.10/site-packages/genesis`. If so, delete it.
Another check is
```python
import genesis as gs
print(gs.__path__) # check if this is as expected
```

arjunsshah, 2024-12-22:
> how did you install genesis?

I am just doing `pip install genesis-world` to install it.

> You can check
> 
> ```shell
> ls $CONDA_PREFIX/genesis/lib/python3.10/site-packages | grep genesis # remember to change the python version accordingly
> ```
> 
> There shouldn't be things like `$CONDA_PREFIX/genesis/lib/python3.10/site-packages/genesis`. If so, delete it. Another check is
> 
> ```python
> import genesis as gs
> print(gs.__path__) # check if this is as expected
> ```

This is what I am getting when I run:
```
import genesis as gs
print(gs.__path__) # check if this is as expected

```
`['/opt/anaconda3/lib/python3.11/site-packages/genesis']
`

I've uninstalled and reinstalled, not too sure whats going on

closed 2024-12-23 by arjunsshah

hongbo-miao, 2025-01-14:
@arjunsshah based on
> ['/opt/anaconda3/lib/python3.11/site-packages/genesis'] 
apparently you installed `genesis` instead of `genesis-world`.

varshneydevansh, 2025-01-28:
I have installed genesis-world in my mac m3 pro 14 inch
https://github.com/Genesis-Embodied-AI/Genesis 
I am seeing no error in import genesis as gs
but when I am doing gs.init()

```py
import genesis as gs

# Define minimal initialization parameters
init_params = {
    'backend': gs.cpu  # or gs.gpu if GPU is available and supported
}

try:
    print("Attempting to initialize Genesis AI...")

    # Initialize the package with the specified parameters
    gs.init(**init_params)

    print("Genesis AI initialized successfully.")
except AttributeError as ae:
    print(f"Attribute error: {ae}")
except ImportError as ie:
    print(f"Import error: {ie}")
except Exception as e:
    print(f"An unexpected error occurred during initialization: {e}")
    import traceback
    traceback.print_exc()
else:
    print("Initialization process completed without errors.")
finally:
    print("Debug session complete.")
```
```sh
Traceback (most recent call last):
  File "<python-input-2>", line 1, in <module>
    import genesis as gs
ModuleNotFoundError: No module named 'genesis'
Attempting to initialize Genesis AI...
An unexpected error occurred during initialization: name 'gs' is not defined
Traceback (most recent call last):
  File "<python-input-2>", line 6, in <module>
    gs.init(backend=gs.cpu)
    ^^
NameError: name 'gs' is not defined
Debug session complete.
```
```bash
(genesis) chiggywiggy@Mac Downloads % pip show genesis-world
Name: genesis-world
Version: 0.2.1
Summary: A universal and generative physics engine
Home-page: 
Author: 
Author-email: 
License: 
Location: /Users/chiggywiggy/genesis/lib/python3.10/site-packages
Requires: black, coacd, freetype-py, libigl, lxml, moviepy, mujoco, numba, numpy, opencv-python, OpenEXR, psutil, pycollada, pydantic, PyGEL3D, pyglet, pygltflib, pymeshlab, PyOpenGL, scikit-image, screeninfo, six, taichi, tetgen
```

I think the better solution would be changing the import statement from genesis to genesis_world.

As pip already have a package named genesis. Me and my friend are stuck in the problem since 3 days.

[image]

varshneydevansh, 2025-01-28:
And yes My friend did installed genesis earlier which we purged from the system. But, this genesis is still having trouble

varshneydevansh, 2025-01-28:
[image]

varshneydevansh, 2025-01-28:
[image]

varshneydevansh, 2025-01-28:
```bash
(genesis) chiggywiggy@Mac Downloads % python3.10 genesis_test.py
Traceback (most recent call last):
  File "/Users/chiggywiggy/Downloads/genesis_test.py", line 1, in <module>
    import genesis as gs
  File "/Users/chiggywiggy/genesis/lib/python3.10/site-packages/genesis/_init_.py", line 2, in <module>
    from unittest.mock import patch
  File "/opt/homebrew/Cellar/python@3.10/3.10.16/Frameworks/Python.framework/Versions/3.10/lib/python3.10/unittest/mock.py", line 26, in <module>
    import asyncio
  File "/opt/homebrew/Cellar/python@3.10/3.10.16/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/_init_.py", line 8, in <module>
    from .base_events import *
  File "/opt/homebrew/Cellar/python@3.10/3.10.16/Frameworks/Python.framework/Versions/3.10/lib/python3.10/asyncio/base_events.py", line 18, in <module>
    import concurrent.futures
  File "/opt/homebrew/Cellar/python@3.10/3.10.16/Frameworks/Python.framework/Versions/3.10/lib/python3.10/concurrent/futures/_init_.py", line 8, in <module>
    from concurrent.futures._base import (FIRST_COMPLETED,
  File "/opt/homebrew/Cellar/python@3.10/3.10.16/Frameworks/Python.framework/Versions/3.10/lib/python3.10/concurrent/futures/_base.py", line 7, in <module>
    import logging
  File "/Users/chiggywiggy/genesis/lib/python3.10/site-packages/Genesis/logging/_init_.py", line 1, in <module>
    from .logger import Logger
  File "/Users/chiggywiggy/genesis/lib/python3.10/site-packages/Genesis/logging/logger.py", line 7, in <module>
    from .time_elapser import TimeElapser
  File "/Users/chiggywiggy/genesis/lib/python3.10/site-packages/Genesis/logging/time_elapser.py", line 6, in <module>
    from genesis.utils.emoji import get_clock
  File "/Users/chiggywiggy/genesis/lib/python3.10/site-packages/genesis/utils/_init_.py", line 1, in <module>
    from .misc import *
  File "/Users/chiggywiggy/genesis/lib/python3.10/site-packages/genesis/utils/misc.py", line 11, in <module>
    import torch
  File "/Users/chiggywiggy/genesis/lib/python3.10/site-packages/torch/_init_.py", line 51, in <module>
    from torch._utils import (
  File "/Users/chiggywiggy/genesis/lib/python3.10/site-packages/torch/_utils.py", line 956, in <module>
    logger = logging.getLogger(_name_)
AttributeError: partially initialized module 'logging' has no attribute 'getLogger' (most likely due to a circular import)
```

varshneydevansh, 2025-01-29:
```bash 
[~]
 devansh  python3                                                                                                                                       [25/01/29|11:07PM]
Python 3.10.12 (main, Jan 17 2025, 14:35:34) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import genesis as gs
>>> gs.init(backend=gs.gpu)
[Genesis] [23:17:00] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [23:17:00] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [23:17:00] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
/home/devansh/.local/lib/python3.10/site-packages/torch/cuda/__init__.py:129: UserWarning: CUDA initialization: Unexpected error from cudaGetDeviceCount(). Did you run some cuda functions before calling NumCudaDevices() that might have already set an error? Error 804: forward compatibility was attempted on non supported HW (Triggered internally at ../c10/cuda/CUDAFunctions.cpp:108.)
  return torch._C._cuda_getDeviceCount() > 0
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/devansh/.local/lib/python3.10/site-packages/genesis/__init__.py", line 82, in init
    device, device_name, total_mem = get_gpu_device()
  File "/home/devansh/.local/lib/python3.10/site-packages/genesis/utils/misc.py", line 117, in get_gpu_device
    gs.raise_exception("cuda device not available")

>>> gs.init(backend=gs.cpu)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/devansh/.local/lib/python3.10/site-packages/genesis/__init__.py", line 50, in init
    raise_exception("Genesis already initialized.")

>>> exit()
[Genesis] [23:29:00] [ERROR] Genesis already initialized.
[Genesis] [23:29:00] [INFO] 💤 Exiting Genesis and caching compiled kernels...
               
 [~]
 devansh  python3 genesis_test.py                                                                                                                       [25/01/29|11:29PM]
Attempting to initialize Genesis AI...
[Genesis] [23:29:20] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [23:29:20] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [23:29:20] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [23:29:20] [INFO] Running on [Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz] with backend gs.cpu. Device memory: 38.99 GB.
[Genesis] [23:29:21] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Genesis AI initialized successfully.
Initialization process completed without errors.
Debug session complete.
[Genesis] [23:29:21] [INFO] 💤 Exiting Genesis and caching compiled kernels...
```

varshneydevansh, 2025-01-29:
```py
import genesis as gs

def initialize_genesis(backend):
    try:
        # Attempt to initialize the package
        gs.init(backend=backend)
        print("Genesis AI initialized successfully.")
    except Exception as e:
        print(f"An error occurred during initialization: {e}")

# Function to check GPU availability
def check_gpu_availability():
    import torch
    return torch.cuda.is_available()

# Main script
try:
    print("Attempting to initialize Genesis AI...")

    # Try GPU first if available
    if check_gpu_availability():
        print("Attempting to initialize Genesis AI with GPU...")
        initialize_genesis(gs.gpu)
    else:
        print("No GPU available. Attempting to initialize Genesis AI with CPU...")
        initialize_genesis(gs.cpu)

    print("Initialization process completed without errors.")
finally:
    print("Debug session complete.")

```
```bash
 ✘  devansh  python3 genesis_test.py                                                                                                                   [25/01/29|11:58PM]
Attempting to initialize Genesis AI...
/home/devansh/.local/lib/python3.10/site-packages/torch/cuda/__init__.py:129: UserWarning: CUDA initialization: Unexpected error from cudaGetDeviceCount(). Did you run some cuda functions before calling NumCudaDevices() that might have already set an error? Error 804: forward compatibility was attempted on non supported HW (Triggered internally at ../c10/cuda/CUDAFunctions.cpp:108.)
  return torch._C._cuda_getDeviceCount() > 0
No GPU available. Attempting to initialize Genesis AI with CPU...
[Genesis] [23:58:30] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [23:58:30] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [23:58:30] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [23:58:30] [INFO] Running on [Intel(R) Core(TM) i7-10750H CPU @ 2.60GHz] with backend gs.cpu. Device memory: 38.99 GB.
[Genesis] [23:58:30] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Genesis AI initialized successfully.
Initialization process completed without errors.
Debug session complete.
[Genesis] [23:58:30] [INFO] 💤 Exiting Genesis and caching compiled kernels...
               
```

varshneydevansh, 2025-01-29:
```py 
devansh  nvcc --version                                                                                        [25/01/30| 1:02AM]
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2021 NVIDIA Corporation
Built on Thu_Nov_18_09:45:30_PST_2021
Cuda compilation tools, release 11.5, V11.5.119
Build cuda_11.5.r11.5/compiler.30672275_0
               
 [~]
 devansh  python3 genesis_test.py                                                                               [25/01/30| 1:06AM]
Attempting to initialize Genesis AI...
Attempting to initialize Genesis AI with GPU...
[Genesis] [01:07:08] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [01:07:08] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [01:07:08] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [01:07:08] [INFO] Running on [NVIDIA GeForce RTX 2060] with backend gs.cuda. Device memory: 5.61 GB.
[Genesis] [01:07:10] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.1, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
Genesis AI initialized successfully.
Initialization process completed without errors.
Debug session complete.
[Genesis] [01:07:10] [INFO] 💤 Exiting Genesis and caching compiled kernels...
               
```

KafuuChikai, 2025-01-30:
Hi @varshneydevansh,

I'm not sure about the issue you're encountering. However, you could try the following steps to see if it resolves the problem:

- Instead of using `pip`, clone the latest repository using `git clone`.
- Create a new Python virtual environment to ensure a clean installation.

varshneydevansh, 2025-01-30:
Okay, will try this on my friends' Mac :)

I didn't put emphasis on this don't know why. But in Linux(Ubuntu) it's now working.

And one more thing if you feel free can reply regarding the access to the generate method as he's working on his Thesis regarding fuel cells.

varshneydevansh, 2025-01-30:
My friend is working on his Master's thesis on Fuel cells and he decided to use the Genesis-World AI for the task of simulation. What do you think we can use for simulations?

We have to analyze the flow in the mesh kinda like how water flows inside leaves.

Gbrothers1, 2025-04-21:
Not sure who this will work for but reinstalling PyTorch into a virtual enviorment worked for me to fix this exact issue. The problem was in the default managed python library manages by brew. Using python -m venv myenv I was able to install PyTorch and components then reinstall genesis from with git clone, following the install directions outlined on the main GH page.