Genesis-Embodied-AI/genesis-world #227: OpenGL error running 'Hello World' code

kayleigh222, 2024-12-22:
Hiya! I'm getting:
 File "/home/<username>/.local/lib/python3.10/site-packages/OpenGL/GL/shaders.py", line 196, in compileProgram
    program.check_validate()
  File "/home/<username>/.local/lib/python3.10/site-packages/OpenGL/GL/shaders.py", line 105, in check_validate
    raise RuntimeError(
RuntimeError: Validation failure (0): b'active samplers with a different type refer to the same texture image unit'

Error when running the Hello World code on CPU:
[image]

Any pointers on how to fix it?

hexadecimal-REBOOT, 2024-12-22:
I'm getting the same issue on a M1 Mac

rghilduta, 2024-12-22:
Seeing the same thing on a rented A10

hexadecimal-REBOOT, 2024-12-22:
Based on the jit_render.py code, the OpenGL shader validation error on M1 Macs is occurring in the texture binding and program compilation process. Here are the key issues and required changes:

1. In the `forward_pass` function, texture units are not properly managed:
```python
def forward_pass(self, renderer, V, P, cam_pos, flags, program_flags, screen_size, color_list=None, reflection_mat=np.identity(4, np.float32), floor_tex=0):
    # Need to track and limit texture units
    active_texture = lighting_texture  # Could exceed M1's 16 texture unit limit
```

2. In the `bind_lighting` function, texture units are not properly bounded:
```python
@njit
def bind_lighting(pid, flags, light, shadow_map, light_matrix, ambient_light, gl):
    active_texture = 0
    # Need to ensure texture unit doesn't exceed 15 (0-15 range for M1)
```

Required changes:

1. Add texture unit management:
```python
MAX_TEXTURE_UNITS = 16  # M1 Mac limit

@njit
def bind_lighting(pid, flags, light, shadow_map, light_matrix, ambient_light, gl):
    active_texture = 0
    # Track used texture units
    used_units = set()
    
    for i in range(n):
        if active_texture >= MAX_TEXTURE_UNITS:
            break
        # Assign texture units ensuring no conflicts
        texture_unit = active_texture % MAX_TEXTURE_UNITS
        used_units.add(texture_unit)
        active_texture += 1
```

2. Modify texture binding in forward pass:
```python
@njit
def forward_pass(...):
    # Track sampler types per texture unit
    sampler_types = {}
    
    for id in idx:
        if active_texture >= MAX_TEXTURE_UNITS:
            break
            
        # Ensure different sampler types use different units
        texture_unit = active_texture % MAX_TEXTURE_UNITS
        if texture_unit not in sampler_types:
            sampler_types[texture_unit] = set()
```

3. Add validation checks:
```python
def _validate_texture_units(self):
    # Ensure total texture units don't exceed M1 limit
    total_units = (
        len(self.shadow_map) +  # Shadow maps
        sum(1 for tex in self.textures if tex[0])  # Material textures
    )
    if total_units > MAX_TEXTURE_UNITS:
        raise ValueError(f"Total texture units ({total_units}) exceeds M1 Mac limit (16)")
```

These changes would:
- Prevent texture unit overflow
- Ensure different sampler types use different units
- Validate texture unit allocation before rendering
- Work within M1 Mac's OpenGL limitations

The error occurs because the current implementation doesn't properly manage texture units within the M1's constraints.

eruffaldi, 2024-12-23:
Same issues on a Windows 11 laptop under WSL2 with both nvidia or intel backend.

aidinism, 2024-12-24:
```
import os
os.environ['PYOPENGL_PLATFORM'] = 'glx'
```

add this before the genesis import

HJWoods, 2024-12-24:
This fixed it for me, also appears to be the solution for #270

closed 2025-02-23 by YilingQiao