Genesis-Embodied-AI/genesis-world #228: Segmentation fault running "hello_genesis.py" on linux

futuyao, 2024-12-22:
I'm running "hello_genesis.py" after installation and got segmentation fault as below.
[image]

My setup is:
- Debian GNU/Linux 11 (bullseye)
- CUDA 11.8
- NVIDIA driver 550.90.07
- Python 3.12.8
- PyTorch 2.5.1
- Conda 24.9.2

futuyao, 2024-12-22:
So I downgraded PyOpenGL from 3.1.7 to 3.1.4, and the seg fault goes away. But now I got this OpenGL error:

```
[Genesis] [07:54:28] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [07:54:28] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [07:54:28] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [07:54:28] [INFO] Running on [Intel(R) Xeon(R) CPU @ 2.20GHz] with backend gs.cpu. Device memory: 31.36 GB.
[Genesis] [07:54:28] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [07:54:28] [INFO] Scene <a203c9f> created.
[Genesis] [07:54:28] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <ee0a147>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.
[Genesis] [07:54:28] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <7d686fc>, morph: <gs.morphs.MJCF(file='/opt/conda/envs/genesis/lib/python3.12/site-packages/genesis/assets/xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [07:54:30] [INFO] Building scene <a203c9f>...
[Genesis] [07:54:35] [INFO] Compiling simulation kernels...
[Genesis] [07:54:43] [INFO] Building visualizer...
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/OpenGL/latebind.py", line 41, in __call__
    return self._finalCall( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/opt/conda/envs/genesis/lib/python3.12/threading.py", line 1075, in _bootstrap_inner
    self.run()
  File "/opt/conda/envs/genesis/lib/python3.12/threading.py", line 1012, in run
    self._target(*self._args, **self._kwargs)
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
           ^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 640, in on_draw
    self._render()
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1086, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 916, in _update_context
    texture._add_to_context()
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/genesis/ext/pyrender/texture.py", line 219, in _add_to_context
    max_aniso = glGetFloat(texture_filter_anisotropic.GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/OpenGL/latebind.py", line 45, in __call__
    return self._finalCall( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/OpenGL/wrapper.py", line 666, in wrapperCall
    raise err
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/OpenGL/wrapper.py", line 659, in wrapperCall
    result = wrappedOperation( *cArguments )
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/OpenGL/platform/baseplatform.py", line 402, in __call__
    return self( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/conda/envs/genesis/lib/python3.12/site-packages/OpenGL/error.py", line 228, in glCheckError
    raise GLError(
OpenGL.error.GLError: GLError(
	err = 1280,
	description = b'invalid enumerant',
	baseOperation = glGetFloatv,
	pyArgs = (
		GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
		<object object at 0x7f2bffc441d0>,
	),
	cArgs = (
		GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
		array([0.], dtype=float32),
	),
	cArguments = (
		GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
		array([0.], dtype=float32),
	)
)
```

duburcqa, 2025-03-23:
Closing because of inactivity.

closed 2025-03-23 by duburcqa