Genesis-Embodied-AI/genesis-world #234: TypeError: Cannot convert a MPS Tensor to float64 dtype as the MPS framework doesn't support float64. Please use float32 instead.

gilons, 2024-12-22:
Hi there, I'm trying to run some examples in the example repository. 
This example `examples/tutorials/sph_liquid.py`

And I got this error

```
Traceback (most recent call last):
  File "/Users/gilesfokam/workspace/personal/Genesis/examples/tutorials/sph_liquid.py", line 29, in <module>
    liquid = scene.add_entity(
             ^^^^^^^^^^^^^^^^^
  File "/Users/gilesfokam/.pyenv/versions/3.12.8/lib/python3.12/site-packages/genesis/utils/misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/gilesfokam/.pyenv/versions/3.12.8/lib/python3.12/site-packages/genesis/engine/scene.py", line 351, in add_entity
    entity = self._sim._add_entity(morph, material, surface, visualize_contact)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/gilesfokam/.pyenv/versions/3.12.8/lib/python3.12/site-packages/genesis/engine/simulator.py", line 140, in _add_entity
    entity = self.sph_solver.add_entity(self.n_entities, material, morph, surface)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/gilesfokam/.pyenv/versions/3.12.8/lib/python3.12/site-packages/genesis/engine/solvers/sph_solver.py", line 140, in add_entity
    entity = SPHEntity(
             ^^^^^^^^^^
  File "/Users/gilesfokam/.pyenv/versions/3.12.8/lib/python3.12/site-packages/genesis/engine/entities/sph_entity.py", line 16, in __init__
    super().__init__(
  File "/Users/gilesfokam/.pyenv/versions/3.12.8/lib/python3.12/site-packages/genesis/engine/entities/particle_entity.py", line 69, in __init__
    self.sample()
  File "/Users/gilesfokam/.pyenv/versions/3.12.8/lib/python3.12/site-packages/genesis/engine/entities/particle_entity.py", line 259, in sample
    self._init_particles_offset = (gs.tensor(particles).contiguous() - gs.tensor(origin)).contiguous()
                                   ^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 3, in tensor
  File "/Users/gilesfokam/.pyenv/versions/3.12.8/lib/python3.12/site-packages/genesis/grad/creation_ops.py", line 64, in torch_op_wrapper
    gs_tensor = from_torch(torch_tensor, dtype, requires_grad, detach=True, scene=scene)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/gilesfokam/.pyenv/versions/3.12.8/lib/python3.12/site-packages/genesis/grad/creation_ops.py", line 92, in from_torch
    gs_tensor = Tensor(torch_tensor.to(gs.device).to(dtype), scene=scene).clone()
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: Cannot convert a MPS Tensor to float64 dtype as the MPS framework doesn't support float64. Please use float32 instead.
```

My system description
[image]

zhouxian, 2024-12-22:
This seems fixed in the main repo. Could you clone the repo and do `pip install -e .`? We will update the pypi version soon

gilons, 2024-12-22:
Okay Nice thanks

closed 2024-12-22 by zhouxian

x4080, 2024-12-22:
Whats the minimum memory needed to run on apple silicon ?

zhouxian, 2024-12-22:
There’s no minimum requirement as far as we know