Genesis-Embodied-AI/genesis-world #235: Cannot Install LuisaRenderPy

Yzhang367, 2024-12-22:
Trying Genesis has really excited me and the greate integrated functions do help a lot in simulation and RL. What made me a little frustrated was that LuisaRender could not run. As a beginner in ray tracing render, I genuinely want to try the LuisaRender in Genesis.

CPU: 13th Gen Intel(R) Core(TM) i9-13980HX
GPU: GeForce RTX 4060 Laptop
Memory: 16GB
OS: Ubuntu24.04

I followed the doc step by step. Everything went well until this step:
```bash
cd genesis/ext/LuisaRender
cmake -S . -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.9 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON # remember to check python version
cmake --build build -j $(nproc)
```
Then warning messages as follows came out:
1.
```bash
CMake Warning at src/compute/scripts/validate_options.cmake:70 (message):
  The DirectX backend is not available.  The DirectX backend will be
  disabled.
Call Stack (most recent call first):
  src/compute/scripts/validate_options.cmake:80 (report_feature_not_available)
  src/compute/CMakeLists.txt:53 (include)

CMake Warning at src/compute/scripts/validate_options.cmake:70 (message):
  The Metal backend is not available.  The Metal backend will be disabled.
Call Stack (most recent call first):
  src/compute/scripts/validate_options.cmake:91 (report_feature_not_available)
  src/compute/CMakeLists.txt:53 (include)

```
But this message might only matter on Windows or Macs.
2.
```bash
-- Found CUDAToolkit: /home/yzhang/.conda/envs/genesis/targets/x86_64-linux/include;/home/yzhang/.conda/envs/genesis/include (found version "12.6.20") 
CMake Warning at src/compute/src/backends/cuda/CMakeLists.txt:149 (message):
  NVTT not found! Please install NVTT from
  https://developer.nvidia.com/nvidia-texture-tools-exporter and set the
  CMake NVTT_DIR variable to the folder containing nvtt*.dll (e.g.
  C:\Program Files\NVIDIA Corporation\NVIDIA Texture Tools) .

CMake Warning at src/compute/src/backends/cuda/CMakeLists.txt:156 (message):
  NVTT_DIR didn't contain an NVTT shared library of the form nvtt*.dll or
  libnvtt.so.*! Is NVTT_DIR set correctly? NVTT_DIR was

CMake Error at src/apps/CMakeLists.txt:12 (find_package):
  By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "pybind11",
  but CMake did not find one.

  Could not find a package configuration file provided by "pybind11" with any
  of the following names:

    pybind11Config.cmake
    pybind11-config.cmake

  Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
  "pybind11_DIR" to a directory containing one of the above files.  If
  "pybind11" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
```
I have tried a two ways to install NVTT but all failed. First, I downloaded the .tar.gz file from the official website but it didn't have a CMakeList.txt. Then I tried to download https://github.com/castano/nvidia-texture-tools and perform cmake install, but I came across errors might relate to CUDA.

I wonder if there was substitution for NVTT or if LuisaRender could run without NVTT.

zswang666, 2024-12-22:
NVTT is warning. The main error is pybind11. Please check [FAQ](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html#faq).
```
pip install -e ".[render]"
```
or
```
pip install "pybind11[global]"
```

OIEIEIO, 2024-12-22:
### Rewritten for my Ubuntu 24 setup

***Note installation path for LuisaRenderPy***

 (genesis2) jorge@jorge-X99:~/Genesis/examples/rendering$ python demo.py --vis

```
import torch
import importlib.util
import genesis as gs
import os

# Function to load LuisaRenderPy dynamically
def load_luisarenderpy():
    luisa_path = "/home/jorge/Genesis/genesis/ext/LuisaRender/build/bin/LuisaRenderPy.cpython-310-x86_64-linux-gnu.so"
    if not os.path.exists(luisa_path):
        raise ImportError(f"LuisaRenderPy not found at {luisa_path}")

    spec = importlib.util.spec_from_file_location("LuisaRenderPy", luisa_path)
    module = importlib.util.module_from_spec(spec)
    spec.loader.exec_module(module)
    print(f"[INFO] Successfully loaded LuisaRenderPy from {luisa_path}")
    return module

# Load LuisaRenderPy
LuisaRenderPy = load_luisarenderpy()

def main():
    ########################## Initialize Genesis ##########################
    gs.init(seed=0, precision="32", logging_level="debug")
    print("[INFO] Genesis initialized successfully.")

    ########################## Create a Scene ##########################
    try:
        scene = gs.Scene(
            sim_options=gs.options.SimOptions(),
            viewer_options=gs.options.ViewerOptions(
                res=(1920, 1080),
                camera_pos=(8.5, 0.0, 4.5),
                camera_lookat=(3.0, 0.0, 0.5),
                camera_fov=50,
            ),
            rigid_options=gs.options.RigidOptions(enable_collision=False, gravity=(0, 0, 0)),
            renderer=gs.renderers.RayTracer(  # type: ignore
                env_surface=gs.surfaces.Emission(
                    emissive_texture=gs.textures.ImageTexture(
                        image_path="textures/indoor_bright.png",
                    ),
                ),
                env_radius=15.0,
                env_euler=(0, 0, 180),
                lights=[
                    {"pos": (0.0, 0.0, 10.0), "radius": 3.0, "color": (15.0, 15.0, 15.0)},
                ],
            ),
        )
        print("[INFO] Scene successfully created.")
    except Exception as e:
        print(f"[ERROR] Failed to create scene: {e}")
        return

    ########################## Add Materials and Entities ##########################
    # Floor
    scene.add_entity(
        morph=gs.morphs.Plane(pos=(0.0, 0.0, -0.5)),
        surface=gs.surfaces.Aluminium(ior=10.0),
    )

    # Various spheres with different materials
    materials = [
        ("meshes/sphere.obj", -3, (1.0, 0.5, 0.5)),
        ("meshes/sphere.obj", -1.8, (1.0, 1.0, 1.0)),
        ("meshes/sphere.obj", -0.6, (0.6, 0.8, 1.0)),
        ("meshes/sphere.obj", 0.6, "Iron"),
        ("meshes/sphere.obj", 1.8, "Gold"),
        ("meshes/sphere.obj", 3.0, "Glass"),
        ("meshes/sphere.obj", -3, (1.0, 1.0, 1.0, 0.5)),
    ]
    for file, pos, material in materials:
        scene.add_entity(
            morph=gs.morphs.Mesh(file=file, scale=0.5, pos=(0.0, pos, 0.0)),
            surface=(
                gs.surfaces.Rough(color=material)
                if isinstance(material, tuple)
                else getattr(gs.surfaces, material)(color=(1.0, 1.0, 1.0))
            ),
        )

    ########################## Add Cameras ##########################
    cam_0 = scene.add_camera(
        res=(1600, 900),
        pos=(8.5, 0.0, 1.5),
        lookat=(3.0, 0.0, 0.7),
        fov=60,
        GUI=True,
        spp=512,
    )

    ########################## Build and Render Scene ##########################
    scene.build()
    print("[INFO] Scene setup complete.")

    horizon = 2000
    scene.reset()
    for i in range(horizon):
        scene.step()
        cam_0.render()

if __name__ == "__main__":
    main()
```
[image]
[image]
[image]

Yzhang367, 2024-12-23:
> NVTT is warning. The main error is pybind11. Please check [FAQ](https://genesis-world.readthedocs.io/en/latest/user_guide/getting_started/visualization.html#faq).
> 
> ```
> pip install -e ".[render]"
> ```
> 
> or
> 
> ```
> pip install "pybind11[global]"
> ```

Thank you very much for your reply! But I have installed all above and the error still existed. When I rebooted and tried to perform cmake installation again, I found out that my 16G memory might be not enough for the installation. Here is the screenshot of my CPU monitor.
[image]

Yzhang367, 2024-12-23:
> ### Rewritten for my Ubuntu 24 setup
> 
> _**Note installation path for LuisaRenderPy**_
> 
> (genesis2) jorge@jorge-X99:~/Genesis/examples/rendering$ python demo.py --vis
> 
> ```
> import torch
> import importlib.util
> import genesis as gs
> import os
> 
> # Function to load LuisaRenderPy dynamically
> def load_luisarenderpy():
>     luisa_path = "/home/jorge/Genesis/genesis/ext/LuisaRender/build/bin/LuisaRenderPy.cpython-310-x86_64-linux-gnu.so"
>     if not os.path.exists(luisa_path):
>         raise ImportError(f"LuisaRenderPy not found at {luisa_path}")
> 
>     spec = importlib.util.spec_from_file_location("LuisaRenderPy", luisa_path)
>     module = importlib.util.module_from_spec(spec)
>     spec.loader.exec_module(module)
>     print(f"[INFO] Successfully loaded LuisaRenderPy from {luisa_path}")
>     return module
> 
> # Load LuisaRenderPy
> LuisaRenderPy = load_luisarenderpy()
> 
> def main():
>     ########################## Initialize Genesis ##########################
>     gs.init(seed=0, precision="32", logging_level="debug")
>     print("[INFO] Genesis initialized successfully.")
> 
>     ########################## Create a Scene ##########################
>     try:
>         scene = gs.Scene(
>             sim_options=gs.options.SimOptions(),
>             viewer_options=gs.options.ViewerOptions(
>                 res=(1920, 1080),
>                 camera_pos=(8.5, 0.0, 4.5),
>                 camera_lookat=(3.0, 0.0, 0.5),
>                 camera_fov=50,
>             ),
>             rigid_options=gs.options.RigidOptions(enable_collision=False, gravity=(0, 0, 0)),
>             renderer=gs.renderers.RayTracer(  # type: ignore
>                 env_surface=gs.surfaces.Emission(
>                     emissive_texture=gs.textures.ImageTexture(
>                         image_path="textures/indoor_bright.png",
>                     ),
>                 ),
>                 env_radius=15.0,
>                 env_euler=(0, 0, 180),
>                 lights=[
>                     {"pos": (0.0, 0.0, 10.0), "radius": 3.0, "color": (15.0, 15.0, 15.0)},
>                 ],
>             ),
>         )
>         print("[INFO] Scene successfully created.")
>     except Exception as e:
>         print(f"[ERROR] Failed to create scene: {e}")
>         return
> 
>     ########################## Add Materials and Entities ##########################
>     # Floor
>     scene.add_entity(
>         morph=gs.morphs.Plane(pos=(0.0, 0.0, -0.5)),
>         surface=gs.surfaces.Aluminium(ior=10.0),
>     )
> 
>     # Various spheres with different materials
>     materials = [
[... 21 lines cut ...]
>         pos=(8.5, 0.0, 1.5),
>         lookat=(3.0, 0.0, 0.7),
>         fov=60,
>         GUI=True,
>         spp=512,
>     )
> 
>     ########################## Build and Render Scene ##########################
>     scene.build()
>     print("[INFO] Scene setup complete.")
> 
>     horizon = 2000
>     scene.reset()
>     for i in range(horizon):
>         scene.step()
>         cam_0.render()
> 
> if __name__ == "__main__":
>     main()
> ```
> 
> ![Screenshot from 2024-12-22 15-01-01](https://private-user-images.githubusercontent.com/47255860/398010203-f14efcfe-fc74-4ad9-b70a-4c799a34f3e5.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ5MjYzNzQsIm5iZiI6MTczNDkyNjA3NCwicGF0aCI6Ii80NzI1NTg2MC8zOTgwMTAyMDMtZjE0ZWZjZmUtZmM3NC00YWQ5LWI3MGEtNGM3OTlhMzRmM2U1LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjIzVDAzNTQzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE0Y2NhY2VhZjlhYWYyMDM5YjBhZDNiZGE4NDM4OThkM2U1MTVjN2Q5ODFmOWQwOGMzN2U5NzVmNDZiZDg1MGUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.zjRBVCnyGkAa9YJNBAtMLuqWInYhOJeLa6VYha2-BAY) ![Screenshot from 2024-12-22 15-00-04](https://private-user-images.githubusercontent.com/47255860/398010212-3973915b-9c8b-4212-bcc3-81dc6dc79d49.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ5MjYzNzQsIm5iZiI6MTczNDkyNjA3NCwicGF0aCI6Ii80NzI1NTg2MC8zOTgwMTAyMTItMzk3MzkxNWItOWM4Yi00MjEyLWJjYzMtODFkYzZkYzc5ZDQ5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjIzVDAzNTQzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ3MWNhNWFjYzlkNDk3MzNmMTIyYmQ2ZmM1MDE4NWJhNmY1ODY0ZWM2OWUxNzY3YzMwMmRiZTAyMGZlZTczODgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.5nH_HOgT4g7G6YeOOli2YnFakEjFleVHejZD44im22k) ![image](https://private-user-images.githubusercontent.com/47255860/398010291-ff348654-5baf-42df-bfec-725fd88d6ce2.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ5MjYzNzQsIm5iZiI6MTczNDkyNjA3NCwicGF0aCI6Ii80NzI1NTg2MC8zOTgwMTAyOTEtZmYzNDg2NTQtNWJhZi00MmRmLWJmZWMtNzI1ZmQ4OGQ2Y2UyLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMjMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjIzVDAzNTQzNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZkMWM1MTJhYjRmN2JkZWQ1Y2QyZjRhNDMxMTIzY2MxMDhmMjc1NGEyOTUzM2QwMmFlOWYxYmQzZGY4ZjZiMjcmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.JvRYatQ3yKwYL5A4aunY0eiad69Ej6CPfX0DEw-CgKY)

Thank you very much for your reply! But I didn't have the chance to try to add the path because the LuisaRenderPy.cpython installation step could not be finished. I found out that my 16G memory might be not enough for the installation. Here is the screenshot of my CPU monitor. What is the memory of your computer? Genuinely thanks for your help!
[image]

zswang666, 2024-12-24:
By installation, you mean compiling Luisa, right? If so, how many processes are you using? Perhaps do single-process make?

Yzhang367, 2024-12-24:
> By installation, you mean compiling Luisa, right? If so, how many processes are you using? Perhaps do single-process make?
> 
> (All the machines I tested have 64G+ cpu memory)

Thank you soooooo much for your help! The problem was perfectly solved when I changed the cmake step as follows:
```bash
cd genesis/ext/LuisaRender
cmake -S. -B build -D CMAKE_BUILD_TYPE=Release -D PYTHON_VERSIONS=3.9 -D LUISA_COMPUTE_DOWNLOAD_NVCOMP=ON # remember to check python version
cmake --build build
```

closed 2024-12-24 by Yzhang367