Genesis-Embodied-AI/genesis-world #236: BUG. Can not visualize texture.

Riften, 2024-12-22:
Genesis can not visualize texture for the MJCF model attached.

```python
import genesis as gs
import os
gs.init(backend=gs.cuda)
scene = gs.Scene(show_viewer=True)
mjcf_morph = gs.morphs.MJCF(file=os.path.join('texture_bug', 'floor_room.xml'))
scene.add_entity(mjcf_morph)
scene.build()
```
Genesis result:

[image]

Mujoco (3.2.5) result:

[image]

MJCF file with texture image:

[texture_bug.zip](https://github.com/user-attachments/files/18221971/texture_bug.zip)

MJCF file
```xml
<mujoco model="base">
  <option impratio="20" cone="elliptic" density="1.2" viscosity="0.00002" timestep="0.002" />
<compiler angle="radian" meshdir="meshes/" inertiagrouprange="0 0" autolimits="true" />
<size nconmax="5000" njmax="5000" />
<asset>
    <material texrepeat="2 2" texuniform="true" reflectance="0.1" shininess="0.1" name="floor_room_wall_mat" texture="floor_room_wall" />
  <texture type="2d" name="floor_room_wall" file="light_wood_planks_long.png" />
  </asset>
<worldbody>
    <body pos="2.15 -2.5 -0.02" name="floor_room_main" quat="0.707 0 0 0.707">
      
    <geom conaffinity="0" contype="0" mass="100" material="floor_room_wall_mat" type="box" size="2.54 3.19 0.02" pos="0 0 0" group="1" name="floor_room_g0_vis" />
    <site type="sphere" group="0" pos="0 0 0" size="0.002 0.002 0.002" rgba="1 0 0 1" name="floor_room_default_site" />
    </body>
  </worldbody>
</mujoco>
```

OIEIEIO, 2024-12-22:
maybe try this example in rendering folder - see if it imports texture like this ? 
[Screencast from 2024-12-22 12-35-40.webm](https://github.com/user-attachments/assets/142f7ddf-cc8f-432a-8205-b799e472885a)

Riften, 2024-12-23:
> maybe try this example in rendering folder - see if it imports texture like this ? [Screencast from 2024-12-22 12-35-40.webm](https://github.com/user-attachments/assets/142f7ddf-cc8f-432a-8205-b799e472885a)

I got the same rendering result as you. The texture is imported fine.

Riften, 2024-12-23:
I notice that the texture is rendered fine when it is defined within `.obj` mesh file. But the texture defined through `<texture>` element within MJCF file is not rendered properly. Here is the rendering result of a kitchen scene from [robocasa](https://robocasa.ai/). There is `ValueError: Incorrect texture coordinate shape` and `ValueError: Cannot reformat 2-channel texture into RGBA` error raised for some objects. I have removed them from scene.

https://github.com/user-attachments/assets/ce67c0ae-96ae-4509-80ad-c0df7c17832b

There are some objects rendered with texture.

The result from mujoco 3.2.5 for the same scene.

[image]

Riften, 2024-12-23:
@zhouxian 抱歉打扰，请问上面提到的问题是 genesis 对 mjcf 的支持问题，还是我对 texture 的导入方式不对？后续有没有修复或特性支持的计划，谢谢。

zhouxian, 2024-12-23:
抱歉我们暂时人手有些不够issue有点多😅一定会修复的！
我们会尽可能支持各种mesh和texture格式。（现在对mjcf的支持应该已经比isaac gym好很多了吧😬

如果过两周还没修复再戳我们一下，感谢！

zhenjia-xu, 2024-12-24:
In #288, box texture is loaded inside mjcf parser. At least the wooden plane looks good.
Please give it a try and let me know if it works.
If so, I will merge it into the main branch.

Riften, 2024-12-24:
> In #288, box texture is loaded inside mjcf parser. At least the wooden plane looks good. Please give it a try and let me know if it works. If so, I will merge it into the main branch.
> 
> BTW, @Riften what's the easiest way to get a few mjcf files of robocasa kitchen environment? I want to load it to see if there is anything missing in our mjcf parser

The floor texture seems to be fine. 

[image]

But there is still some error when load some other texture. The mjcf file which raises error is attached.

[texture_bug_2.zip](https://github.com/user-attachments/files/18238794/texture_bug_2.zip)

<details>
<summary>Full error message</summary>

```bash
{
	"name": "TypeError",
	"message": "Cannot handle this data type: (1, 1, 600), |u1",
	"stack": "---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
File ~/App/anaconda3/envs/robotic/lib/python3.9/site-packages/PIL/Image.py:3098, in fromarray(obj, mode)
   3097 try:
-> 3098     mode, rawmode = _fromarray_typemap[typekey]
   3099 except KeyError as e:

KeyError: ((1, 1, 600), '|u1')

The above exception was the direct cause of the following exception:

TypeError                                 Traceback (most recent call last)
Cell In[5], line 2
      1 #mjcf_entity = scene.add_entity(mjcf_morph, surface=file_surface)
----> 2 mjcf_entity = scene.add_entity(mjcf_morph)
      3 #obj_entity = scene.add_entity(obj_morph)

File ~/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/utils/misc.py:38, in assert_unbuilt.<locals>.wrapper(self, *args, **kwargs)
     36 if self.is_built:
     37     gs.raise_exception(\"Scene is already built.\")
---> 38 return method(self, *args, **kwargs)

File ~/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/engine/scene.py:360, in Scene.add_entity(self, morph, material, surface, visualize_contact, vis_mode)
    357     else:
    358         morph.decompose_nonconvex = False
--> 360 entity = self._sim._add_entity(morph, material, surface, visualize_contact)
    362 return entity

File ~/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/engine/simulator.py:134, in Simulator._add_entity(self, morph, material, surface, visualize_contact)
    131     entity = self.avatar_solver.add_entity(self.n_entities, material, morph, surface, visualize_contact)
    133 elif isinstance(material, gs.materials.Rigid):
--> 134     entity = self.rigid_solver.add_entity(self.n_entities, material, morph, surface, visualize_contact)
    136 elif isinstance(material, gs.materials.MPM.Base):
    137     entity = self.mpm_solver.add_entity(self.n_entities, material, morph, surface)

File ~/App/anaconda3/envs/robotic/lib/python3.9/site-packages/genesis/engine/solvers/rigid/rigid_solver_decomp.py:63, in RigidSolver.add_entity(self, idx, material, morph, surface, visualize_contact)
     60     else:
     61         entity_class = RigidEntity
---> 63 entity = entity_class(
     64     scene=self._scene,
     65     solver=self,
     66     material=material,
     67     morph=morph,
     68     surface=surface,
     69     idx=idx,
     70     idx_in_solver=self.n_entities,
     71     link_start=self.n_links,
     72     joint_start=self.n_joints,
     73     q_start=self.n_qs,
     74     dof_start=self.n_dofs,
     75     geom_start=self.n_geoms,
[... 287 lines cut ...]
    asset_element = root.find('asset')
    header = MJCFHeader(env_xml_str)
    asset = MJCFAsset(ET.tostring(asset_element))
    
    for fixture_name in env.fixtures:
        if fixture_name in _UNSUPPORTED_FIXTURES:
            print(f"Skipping {fixture_name} due to {_UNSUPPORTED_FIXTURES[fixture_name]}")
            continue
        fixture = env.fixtures[fixture_name]
        fixture_mjcf = build_mjcf_from_body_elements([fixture.get_obj()], header, asset, 
                                                    remove_group_0 = False)
        set_group_transparency(fixture_mjcf, transparent_group, alpha=0.0)
        with open(os.path.join(out_dir, f'{fixture_name}.xml'), 'w') as f:
            f.write(ET.tostring(indent(fixture_mjcf), encoding='unicode'))
            
            
if __name__ == '__main__':
    env = load_robocasa_env(1,1)
    os.makedirs('fixtures', exist_ok=True)
    export_fixtures_to_dir(env, 'fixtures')
```

</details>

Feel free to use it.

zhenjia-xu, 2024-12-24:
I fixed PR #288 to support gray images as textures. Please give it a try.
Thanks for providing the code. I randomly loaded 5 fixtures, and it looks good to me. Let me know if there are any further issues.

Riften, 2024-12-25:
> I fixed PR #288 to support gray images as textures. Please give it a try. Thanks for providing the code. I randomly loaded 5 fixtures, and it looks good to me. Let me know if there are any further issues.

Textures are loaded, while the position and scale of geoms seems to be wrong.

before

https://github.com/user-attachments/assets/15752613-e33f-4b1b-90fa-f6eb90cf30b7 

now

https://github.com/user-attachments/assets/688786bc-bb7d-44a6-9920-0d0114cc31d9

I just load all the fixtures exported.

```python
def load_mjcf_dir(scene: gs.Scene, load_dir: str):
    for mjcf_filename in os.listdir(load_dir):
        if not mjcf_filename.endswith('.xml'):
            continue
        mjcf_morph = gs.morphs.MJCF(file=os.path.join(load_dir, mjcf_filename))
        scene.add_entity(mjcf_morph)
```

zhenjia-xu, 2024-12-27:
Hi Riften, the bug has been fixed #288.
Sorry for the inconvenience.

Riften, 2024-12-27:
It works! Thank you very much for the timely fix! I will close this issue shortly.

[image]

closed 2024-12-27 by Riften

zhenjia-xu, 2024-12-27:
great! I will merge into main.