Genesis-Embodied-AI/genesis-world #237: AttributeError: function 'glGetUniformLocation' not found

FazedWithPhase, 2024-12-22:
I tried to run this code:
[image]
and this error happened:
AttributeError: function 'glGetUniformLocation' not found

How do I fix this?

Ed-Peterson, 2024-12-22:
OpenGL version: 4.6.0 NVIDIA 566.36
GPU: GTX 1060 6GB
Error messages: glGetUniformLocation and capture_frame

IaMaAVerYRANdOmPerSoN, 2024-12-22:
[Genesis] [14:28:01] [INFO] ╭─────────────────────────────────────────────────────────────────────────────────────╮
[Genesis] [14:28:01] [INFO] │┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉ Genesis ┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉┈┉│
[Genesis] [14:28:01] [INFO] ╰─────────────────────────────────────────────────────────────────────────────────────╯
[Genesis] [14:28:01] [INFO] Running on [NVIDIA GeForce RTX 3060] with backend gs.cuda. Device memory: 12.00 GB.
[Genesis] [14:28:02] [INFO] 🚀 Genesis initialized. 🔖 version: 0.2.0, 🌱 seed: None, 📏 precision: '32', 🐛 debug: False, 🎨 theme: 'dark'.
[Genesis] [14:28:02] [INFO] Scene <ee13b6d> created.
[Genesis] [14:28:02] [INFO] Adding <gs.RigidEntity>. idx: 0, uid: <f778d5b>, morph: <gs.morphs.Plane>, material: <gs.materials.Rigid>.    
[Genesis] [14:28:02] [INFO] Adding <gs.RigidEntity>. idx: 1, uid: <000d928>, morph: <gs.morphs.MJCF(file='C:\Users\wenmi\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\genesis\assets\xml/franka_emika_panda/panda.xml')>, material: <gs.materials.Rigid>.
[Genesis] [14:28:03] [INFO] Building scene <ee13b6d>...
[Genesis] [14:28:19] [INFO] Compiling simulation kernels...
[Genesis] [14:28:39] [INFO] Building visualizer...
Traceback (most recent call last):
  File "c:\Users\wenmi\Hello_Genesis.py", line 34, in <module>
    scene.build()
  File "C:\Users\wenmi\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\wenmi\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\genesis\engine\scene.py", line 556, in build
    self._visualizer.build()
  File "C:\Users\wenmi\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\genesis\vis\visualizer.py", line 101, in build
    self._context.build(self._scene)
  File "C:\Users\wenmi\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\genesis\vis\rasterizer_context.py", line 70, in build
    self.jit = JITRenderer(self._scene, [], [])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\wenmi\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\genesis\ext\pyrender\jit_render.py", line 201, in __init__
    self.gen_func_ptr()
  File "C:\Users\wenmi\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\genesis\ext\pyrender\jit_render.py", line 354, in gen_func_ptr
    self.gl = GLWrapper()
              ^^^^^^^^^^^
  File "C:\Users\wenmi\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 24, in __init__
    load_func("glGetUniformLocation", GLint, GLuint, GLvoidp)
  File "C:\Users\wenmi\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\genesis\ext\pyrender\numba_gl_wrapper.py", line 52, in load_func
    func_ptr = GL.platform.ctypesloader.buildFunction(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\wenmi\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.12_qbz5n2kfra8p0\LocalCache\local-packages\Python312\site-packages\OpenGL\platform\ctypesloader.py", line 96, in buildFunction
    return functionType( (name, dll), )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: function 'glGetUniformLocation' not found
[Genesis] [14:28:40] [INFO] 💤 Exiting Genesis and caching compiled kernels...

L42ARO, 2024-12-23:
I have the same issue, any help?

achaosss, 2024-12-23:
I have the same issue, any help?

achaosss, 2024-12-23:
I can use by below, set show_viewer=False
scene = gs.Scene(show_viewer=False)

Stoneslf, 2024-12-24:
Windows 11 Family Chinese Version 24H2
Operating system version: 26100.2605
OpenGL version: 4.6.0 NVIDIA 552.22
GPU: NVIDIA GeForce RTX 4050 Laptop GPU 
Python: 3.9.21
cudatoolkit: 11.8.0; cudnn: 8.9.2.26; torch: 2.5.1+cu118; torchaudio: 2.5.1+cu118; torchvision: 0.20.1+cu118

**code:**
```
import genesis as gs
gs.init(backend=gs.cpu)

scene = gs.Scene(show_viewer=True)
plane = scene.add_entity(gs.morphs.Plane())
franka = scene.add_entity(
    gs.morphs.MJCF(file='xml/franka_emika_panda/panda.xml'),
)

scene.build()

for i in range(1000):
    scene.step()
```
**Error messages:** 
[image]

Stoneslf, 2024-12-24:
There are unofficial solutions mentioned in #12 , but some also point out that they may not work well. Overall, Windows is not suitable for using Genesis. I gave up the idea of using it on Windows.

ziyanx02, 2024-12-27:
Close as a duplicate of #12.

closed 2024-12-27 by ziyanx02