Genesis-Embodied-AI/genesis-world #242: [ERROR] Camera not yet supported on Windows but is supposed to. Can anyone help?

BruskB, 2024-12-22:
Hey, when I try to run Genesis locally it gives me the error:

Traceback (most recent call last):
  File "...\Python\GENESIS\test.py", line 32, in <module>
    cam = scene.add_camera(
  File "...\Python\GENESIS\lib\site-packages\genesis\utils\misc.py", line 38, in wrapper
    return method(self, *args, **kwargs)
  File "...\Python\GENESIS\lib\site-packages\genesis\engine\scene.py", line 444, in add_camera
    return self._visualizer.add_camera(res, pos, lookat, up, model, fov, aperture, focus_dist, GUI, spp, denoise)
  File "...\Python\GENESIS\lib\site-packages\genesis\vis\visualizer.py", line 72, in add_camera
    gs.raise_exception("Camera not yet supported on Windows but is supposed to. Can anyone help?")

[Genesis] [21:02:58] [ERROR] Camera not yet supported on Windows but is supposed to. Can anyone help?
[Genesis] [21:02:58] [INFO] 💤 Exiting Genesis and caching compiled kernels...

Am I doing something wrong? What could be the cause of this issue?

IaMaAVerYRANdOmPerSoN, 2024-12-22:
[Genesis] [21:02:58] [ERROR] **Camera not yet supported on Windows but is supposed to. Can anyone help?**

IaMaAVerYRANdOmPerSoN, 2024-12-22:
** Camera not yet supported on Windows but is supposed to. Can anyone help?**

zhouxian, 2024-12-22:
Are you using windows? then it's what the error message is saying...

in our documentation we stated that rendering is not supported on windows, so we would love if someone could help!

BruskB, 2024-12-22:
Ah my bad I was just reading the Documentation, I will run it on Linux instead. I'll look up the code to see what causes the problem not sure if I can help but will try to understand the reasoning behind it at least, I been running some other test codes to see if I can make it work but seems like a workaround will not be efficient for sure and the code needs to be optimized for windows

BruskB, 2024-12-22:
but except for that, it is an incredible work!

closed 2024-12-22 by BruskB