Genesis-Embodied-AI/genesis-world #243: conflicting dependencies.

catv1, 2024-12-22:
while installing it I got this error

(trs) C:\Users\crims>pip install genesis-world
Collecting genesis-world
  Using cached genesis_world-0.2.0-py3-none-any.whl.metadata (11 kB)
Collecting psutil (from genesis-world)
  Using cached psutil-6.1.1-cp37-abi3-win_amd64.whl.metadata (23 kB)
Collecting scikit-image (from genesis-world)
  Using cached scikit_image-0.25.0-cp313-cp313-win_amd64.whl.metadata (14 kB)
INFO: pip is looking at multiple versions of genesis-world to determine which version is compatible with other requirements. This could take a while.
Collecting genesis-world
  Using cached genesis_world-0.1.1-py3-none-any.whl.metadata (11 kB)
ERROR: Cannot install genesis-world==0.1.1 and genesis-world==0.2.0 because these package versions have conflicting dependencies.

The conflict is caused by:
    genesis-world 0.2.0 depends on taichi==1.7.2
    genesis-world 0.1.1 depends on taichi==1.7.2

To fix this you could try to:
1. loosen the range of package versions you've specified
2. remove package versions to allow pip to attempt to solve the dependency conflict

ERROR: ResolutionImpossible: for help visit https://pip.pypa.io/en/latest/topics/dependency-resolution/#dealing-with-dependency-conflicts

(trs) C:\Users\crims>

Ethanote, 2024-12-25:
May I kindly ask how do you solve the problem since I encountered same problem with you

catv1, 2025-01-01:
I never fixed it I just gave up.

YilingQiao, 2025-02-23:
Sorry for the confusion. It seems that PyPI downloaded both 0.1.1 and 0.2.0—strange. You could install from a local folder using `pip install -e .`, or the new 0.2.1 should work fine.

closed 2025-02-23 by YilingQiao