Genesis-Embodied-AI/genesis-world #245: General training support

yic03685, 2024-12-22:
Hi, will there be supports for non-robot motion? For example, besides the character motion provided by the framework, will there be training function to support custom motions?

zhouxian, 2024-12-23:
Could you provide a reference/example explaining what you mean in details?

yic03685, 2024-12-23:
Sure and apologizing for my lack of context. I’m confused how the character motions in the demo were trained? Were all motions trained by individually IK training and used to train the generative model? Will there be support for training custom motions? I understand the generative component still work in progress but if there’s some light document or diagram explaining the connection or IO between the generative and the physic engines, it’d be very helpful. Thanks

ziyanx02, 2024-12-27:
Here is the paper that may help you: 
https://frank-zy-dou.github.io/projects/CASE/index.html

closed 2024-12-27 by ziyanx02