Genesis-Embodied-AI/genesis-world #246: WSL2 OpenGL Error

TalalSultan, 2024-12-22:
i have downloaded genesis on a windows 11 WSL2 to try it out , but when i run hello_genesis.py or any other i get this error
```
MESA: error: ZINK: failed to choose pdev
glx: failed to create drisw screen
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/usr/lib/python3.12/threading.py", line 1073, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.12/threading.py", line 1010, in run
    self._target(*self._args, **self._kwargs)
  File "/home/talal/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/talal/lib/python3.12/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/talal/lib/python3.12/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/talal/lib/python3.12/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/home/talal/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/home/talal/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 640, in on_draw
    self._render()
  File "/home/talal/lib/python3.12/site-packages/genesis/ext/pyrender/viewer.py", line 1086, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/talal/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/home/talal/lib/python3.12/site-packages/genesis/ext/pyrender/renderer.py", line 899, in _update_context
    p._add_to_context()
  File "/home/talal/lib/python3.12/site-packages/genesis/ext/pyrender/primitive.py", line 359, in _add_to_context
    glVertexAttribPointer(0, 3, GL_FLOAT, GL_FALSE, FLOAT_SZ * 3, ctypes.c_void_p(0))
  File "/home/talal/lib/python3.12/site-packages/OpenGL/latebind.py", line 63, in __call__
    return self.wrapperFunction( self.baseFunction, *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/talal/lib/python3.12/site-packages/OpenGL/GL/VERSION/GL_2_0.py", line 469, in glVertexAttribPointer
    contextdata.setValue( key, array )
  File "/home/talal/lib/python3.12/site-packages/OpenGL/contextdata.py", line 58, in setValue
    context = getContext( context )
              ^^^^^^^^^^^^^^^^^^^^^
  File "/home/talal/lib/python3.12/site-packages/OpenGL/contextdata.py", line 40, in getContext
    raise error.Error(
OpenGL.error.Error: Attempt to retrieve context when no valid context
```

yliu000, 2024-12-23:
import os
os.environ['PYOPENGL_PLATFORM'] = 'glx'

I met similiar problem. And by setting PYOPENGL_PLATFORM to 'glx', can work around.

kayleigh222, 2024-12-23:
@yliu000 which file are you adding this to?

yliu000, 2024-12-23:
@kayleigh222 in the main file, before any other import.

I guess you can also do "export PYOPENGL_PLATFORM=glx" in bash.

kayleigh222, 2024-12-23:
legend that works, thank you @yliu000! 

if it helps anyone else, i found [ this issue ](https://github.com/Genesis-Embodied-AI/Genesis/issues/37) has more fixes for the same problem

TalalSultan, 2024-12-23:
Worked for me thanks 

> @kayleigh222 in the main file, before any other import.
> 
> I guess you can also do "export PYOPENGL_PLATFORM=glx" in bash.

TalalSultan, 2024-12-23:
> @kayleigh222 in the main file, before any other import.
> 
> I guess you can also do "export PYOPENGL_PLATFORM=glx" in bash.

closed 2024-12-23 by TalalSultan

laotanlaoban, 2024-12-23:
export PYOPENGL_PLATFORM=glx  对我没有用，有人能解决吗