Genesis-Embodied-AI/genesis-world #249: OpenGL.error

PeiZhangNEU, 2024-12-23:
[Genesis] [08:27:07] [INFO] Building visualizer...
Exception in thread Thread-2 (_init_and_start_app):
Traceback (most recent call last):
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/OpenGL/latebind.py", line 43, in __call__
    return self._finalCall( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: 'NoneType' object is not callable

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/threading.py", line 1045, in _bootstrap_inner
    self.run()
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/threading.py", line 982, in run
    self._target(*self._args, **self._kwargs)
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 1149, in _init_and_start_app
    pyglet.clock.tick()
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/pyglet/clock.py", line 528, in tick
    return _default.tick(poll)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/pyglet/clock.py", line 270, in tick
    self.call_scheduled_functions(delta_t)
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/pyglet/clock.py", line 217, in call_scheduled_functions
    item.func(now - item.last_ts, *item.args, **item.kwargs)
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 941, in _time_event
    self.on_draw()
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 640, in on_draw
    self._render()
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/genesis/ext/pyrender/viewer.py", line 1086, in _render
    retval = renderer.render(self.scene, flags, seg_node_map=seg_node_map)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/genesis/ext/pyrender/renderer.py", line 141, in render
    self._update_context(scene, flags)
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/genesis/ext/pyrender/renderer.py", line 916, in _update_context
    texture._add_to_context()
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/genesis/ext/pyrender/texture.py", line 219, in _add_to_context
    max_aniso = glGetFloat(texture_filter_anisotropic.GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/OpenGL/latebind.py", line 47, in __call__
    return self._finalCall( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/OpenGL/wrapper.py", line 678, in wrapperCall
    raise err
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/OpenGL/wrapper.py", line 671, in wrapperCall
    result = wrappedOperation( *cArguments )
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/OpenGL/platform/baseplatform.py", line 415, in __call__
    return self( *args, **named )
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/zp/anaconda3/envs/genesis/lib/python3.11/site-packages/OpenGL/error.py", line 230, in glCheckError
    raise self._errorClass(
OpenGL.error.GLError: GLError(
        err = 1280,
        description = b'invalid enumerant',
        baseOperation = glGetFloatv,
        pyArgs = (
                GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
                <object object at 0x7f5f45aa3f30>,
        ),
        cArgs = (
                GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
                array([0.], dtype=float32),
        ),
        cArguments = (
                GL_MAX_TEXTURE_MAX_ANISOTROPY_EXT,
                array([0.], dtype=float32),
        )
)

RTX3090
NVIDIA-SMI 555.42.02
Driver Version: 555.42.02
CUDA Version: 12.5
Python: 3.11.11

chenfei8888, 2024-12-23:
I have a same error

zhang-qiang-github, 2024-12-23:
I have a same error

DoiryCool, 2024-12-24:
I have a same error

sridhar-mani, 2024-12-26:
any solution

VerizonCao, 2024-12-27:
same issue on Mac M3.

z-yf17, 2024-12-30:
same issue

sridhar-mani, 2024-12-31:
i found a work around is the bug still open

DoiryCool, 2024-12-31:
> i found a work around is the bug still open

so what’s the problem?

sridhar-mani, 2024-12-31:
the problem is the opengl obviously just change pyopengl version and override mesa

closed 2025-04-25 by duburcqa